SeisComP3/seiscomp3 #10: the fdsnws service skips pre-amp response stages but keeps the correct sensitivity

ozym, 2015-01-30:
We have a few sites where a pre-amp analogue stage is added after the sensor but before the datalogger. The dataless seed when loaded in via dlsv2inv shows the stage in the xml output, however this stage is missing when a web query is made ... i.e.

an example dataless volume: 

ftp://ftp.geonet.org.nz/seed/SINGLE/nz.dataless.seed.epaz

[which has a per-amp gain of 30]

and the resultant fdsnws query:

http://service.geonet.org.nz/fdsnws/station/1/query?station=EPAZ&channel=EHZ&level=response

which has only 2 of the stages returned.  Applications that check the gains (i.e. obspy) complain about a mismatch with the sensitivity value which is causing some concern for our data users.

ozym, 2015-01-30:
I'm seeing a log message, indicating a missing PAZ. response. 
```
Jan 31 00:24:07 avdas03 fdsnws: [warning/FDSNXML] PAZ response not found in inventory: ResponsePAZ#20140429222332.078544.9295
Jan 31 00:24:07 avdas03 fdsnws: [warning/FDSNXML] Stopping at response stage 2
Jan 31 00:24:07 avdas03 fdsnws: [notice/log] fdsnws-station: returned 1Net, 1Sta, 1Loc, 1Cha, 1DL, 1Dec, 1Sen, 1Res (total objects/bytes: 8/3969)
```

ozym, 2015-01-30:
I've reloaded the dataless XML and have these sections:

```
    <datalogger publicID="Datalogger#20150130114214.785947.10428" name="EPAZ.2011.235.EHZ10">
      <description>EPAZ.2011.235.EHZ10</description>
      <gain>419430</gain>
      <maxClockDrift>0.01</maxClockDrift>
      <calibration serialNumber="xxxx" channel="0">
        <start>2011-08-23T00:01:00.0000Z</start>
        <gain>419430</gain>
        <gainFrequency>15</gainFrequency>
      </calibration>
      <decimation sampleRateNumerator="100" sampleRateDenominator="1">
        <analogueFilterChain>ResponsePAZ#20150130114214.786503.10430</analogueFilterChain>
        <digitalFilterChain>ResponseFIR#20150130114214.786184.10429</digitalFilterChain>
      </decimation>
    </datalogger>
```
and
```
    <responsePAZ publicID="ResponsePAZ#20150130114214.786503.10430" name="EPAZ.10.EHZ.2011.235.stage_2">
      <type>A</type>
      <gain>30</gain>
      <gainFrequency>15</gainFrequency>
      <normalizationFactor>1</normalizationFactor>
      <normalizationFrequency>15</normalizationFrequency>
      <numberOfZeros>0</numberOfZeros>
      <numberOfPoles>0</numberOfPoles>
      <zeros/>
      <poles/>
    </responsePAZ>
```

but I still see error messages of:

```
Jan 31 01:18:59 avdas03 fdsnws: [notice/log] loading inventory
Jan 31 01:19:06 avdas03 fdsnws: [warning/FDSNXML] PAZ response not found in inventory: ResponsePAZ#20150130114214.786503.10430
Jan 31 01:19:06 avdas03 fdsnws: [warning/FDSNXML] Stopping at response stage 2
Jan 31 01:19:06 avdas03 fdsnws: [notice/log] fdsnws-station: returned 1Net, 1Sta, 1Loc, 1Cha, 1DL, 1Dec, 1Sen, 1Res (total objects/bytes: 8/2702)
```

gempa-stephan, 2015-01-30:
I think I found the issue. Can you please try the attached patch.

ozym, 2015-01-30:
That's fixed the missing PAZ stage!, However it now looks like they are coming in out of order. 

i.e. now we get  M/S -> VOLTS,  VOLTS -> COUNTS, VOLTS -> VOLTS, COUNTS -> COUNTS etc.

looking at  `Convert2StaXML::process` in `convert2staxml.cpp` shows that `sx_resp0` is setup first and then the analogue and digital stages are processed.

Tricky bit being that this would be fine if the analogue stages (as per how they are defined) have COUNTS as input / output but not if the are in VOLTS (or input units).

Perhaps using the input units as a test for whether the analogue stage should be before the digitisation stage or after may be an option.

gempa-jabe, 2015-02-04:
> ftp://ftp.geonet.org.nz/seed/SINGLE/nz.dataless.seed.epaz

The link is down. Can you upload or send the file again?

gempa-jabe, 2015-02-04:
@ozym please check the above fix if that works for you. I tested with another example and it seems to look right now.

gempa-jabe, 2015-02-04:
> The link is down. Can you upload or send the file again?

I got the file ...

ozym, 2015-02-05:
Great, thanks Jan. From what I can see in the newly built xml the two changes have fixed the problem (tested on centos7). I'll update this issue when I get some feedback from the service users who were having problems.

ozym, 2015-02-05:
User feedback has been extremely positive! Looks like this has been resolved. Ta.

gempa-jabe, 2015-02-05:
My test cases look also OK. Code is merged.

closed 2015-02-05 by gempa-jabe