SeisComP3/seiscomp3 #13: scolv magnitude tab enhancements - support (de)selection of multiple stations

jfclinton, 2015-02-09:
In scolv, on the magnitude tab, allow selection / deselection of multiple stations based on e.g. channel type or distance.

gempa-jabe, 2015-02-17:
I implemented a version you can try. 4 new buttons/actions were added to the magnitude view:
- select a particular subset of rows (shortcut: s)
- edit the selection filter (shortcut: shift+s)
- activate the selected rows (shortcut: +)
- deactivate the selected rows (shortcut: -)

[image]

The selection filter is by default unset and needs to be specified. You can select a column you want to filter, an operation and a value, e.g. distance < 100. The currently set filter is applied again when just pressing `s` next time.
To change the filter use `shift+s`. The result of the selection are a set of selected rows. Nothing has happened yet apart from the selection.
To activate the selected rows, press `+`. To deactivate them use `-`. This also works if you select rows manually as usual.

jfclinton, 2015-02-20:
This is already a satisfactory solution but for the SED use case it could be improved.
We will use it in 2 ways, using Mlh for magnitude:
1 (most important) - filter out strong motion channels. So ideally, we need to support wildcards in the selection filter: <HG?, HN?> for all combinations of HGN, HGE, HG2, HG3, HNN, HNZ, HN2, HN3. I didn't see yet that wildcards, or indeed the 'like' option, supported this flexibility - but maybe I use incorrect syntax
2 (less important) - filter out stations beyond a certain distance

I note:
- Its nice that for each session, the most recent selection filter is remembered, but it is sub-optimal we need to always recall the filter each time we start a new session
- Ideally we would be able to configure preferred filters and have them applied very easily. In the SED case, this would be 2 filters settings as above.

gempa-jabe, 2015-02-20:
> indeed the 'like' option, supported this flexibility - but maybe I use incorrect syntax

In your case **like** is correct and the value should be **HG?** or **HG***. What is currently not configurable via the GUI are logical combinations such as **or** or **and**.

> Ideally we would be able to configure preferred filters and have them applied very easily

OK. The current version is more of a test to check if the general idea fits with what you want. Configuration in .cfg files and logical operations are a bit more complicated ...

jfclinton, 2015-04-15:
Hi Jan,

we use this implementation, and are happy with it. We would like to see the following improvements:

- offer in the window additional AND option so we can combine e.g.
Channel like HG?
AND
Channel like HN?
AND
Distance less / equal 15

- allow similar options in config., so this can be systematically replicated in each session by each user. 

But if this is a major effort (>4 hours), it is not worth it. With the option to select and remove:
"Channel like HG?"
I already have the most important functionality.

thanks,

John

gempa-jabe, 2015-04-16:
Hi John, implemented with 4d5a8e76d5f4e6d0ca526678558f46d1d3e76c94. It allows to add more filters and the current settings are persistent over scolv restarts. I have decided not to store the settings to scolv.cfg but to the Qt settings which live in ~/.config/[agency]/scolv.conf.

It would look like this:
```
...
[MagnitudeView]
selectionFilter="Loc/Cha,Equal,BHZ"
...
```

The Qt setttings also store the last window position and other stuff that is restored when starting scolv again.

gempa-jabe, 2015-04-16:
In the dialog where you configure the filter are now two additional buttons in the upper right corner to add a filter and to remove the last filter line. Let me know if that is OK for you.

gempa-jabe, 2015-07-02:
Any updates? Can I close this issue?

closed 2015-07-23 by gempa-jabe