SeisComP3/seiscomp3 #14: Enhanced filter toggling in scolv picker window

jfclinton, 2015-02-09:
Currently, in scolv (and other guis) waveform windows, the filter can only be toggled on-off using the F hotkey, using the currently selected filter. Can filter toggling be made more sophisticated, where pressing 'F' supports cycling through a number of predefined filters. This would support more rapid review of network that include strong motion - a narrow filter can be rapidly applied w/ and w/o integration.

Example application (filter 1 on (bp1-20Hz); filter 2 on (bp1-20Hz + INT); filter off)

gempa-jabe, 2015-02-09:
It would be quite simple to achieve that in e.g. scrttv. In scolv we need to figure out how to define the cycle sequence. We have already a list of filters from which you can select the current. Should this feature use that list or another one? Is integration just an example or the only use case? Otherwise we could add an option to integrate your data. Cycling through the current list of scolv filters via hotkey is nonetheless a good addition.

jsaul, 2015-02-09:
I also support the idea. In scolv, it might be achieved through the 
current filter selection menu e.g. by adding a checkbox to every entry. 
Then only the selected filters are considered when toggling using key 
"F". If that is technically feasible.

In the scolv.cfg filter config we can prefix an entry with "@", which 
means "default filter". Perhaps this can be adopted any any enty with 
"@" is used for toggling by default. However, some mechanism for 
selecting the toggle traces at runtime is needed anyway.

gempa-jabe, 2015-02-11:
I added a test for such a feature with faaeceede804490805c423ea0fa4793e728315c7 to scrttv. You can define a list of filters with e.g.:

```
filters = "BW(4,0.7,2)", "BW(4,2,4)"
```

and use `f` and `shift+f` to cycle through them.

If that is what you want then the same should be accomplished in scolv with `picker.filters` but only on a subset of them. We could prepend a character as Joachim suggested and cycle through that list.

An open question is about the shortcuts. We can toggle filtering on/off already with `f`. `shift+f` toggles all/current trace. We could use `ctrl+f` to cycle and `ctrl+shift+f` to cycle reverse.

aemanov, 2015-02-12:
For scolv I would suggest to leave F for toggling filtering on/off and use other keys to go through filter list like C - forward, D - backward.

gempa-jabe, 2015-02-13:
@aemanov I agree, and while thinking about it why to define a subset to cycle through? If we can cycle forward and backward at each filter list position it is up to the user to define a well ordered filter list.

I think I understand @jfclinton better now: instead of having

```
picker.filters = "BP 0.1 - 1 Hz  3rd order;BW(3,0.1,1)",
                 "BP 0.1 - 2 Hz  3rd order;BW(3,0.1,2)",
                 ...
```

he would prefer

```
picker.filters = "BP 0.1 - 1 Hz  3rd order;BW(3,0.1,1);BP 0.1 - 1 Hz  3rd order + INT;BW(3,0.1,1)>>INT",
                 "BP 0.1 - 2 Hz  3rd order;BW(3,0.1,2)",
                 ...
```

That means that a filter in the list has actually more than one representation that can be cycled through, kind of 2 dimensional list. Is that correct?

jfclinton, 2015-02-13:
Jan, I'm happy with this solution - I can easily order my list so the relevant pairs I want to switch between across are always together.

gempa-jabe, 2015-02-16:
OK, so we agree on adding two new shortcuts to activate the previous/next filter in the list. The proposal from @aemanov sounds reasonable, but I prefer 'D' for next filter and 'C' for previous filter because 'D' is above 'C' on the keyboard which supports the semantics of up/next.

gempa-jabe, 2015-02-17:
It is implemented but shortcuts had to change because `c` is already used for toggling clobber traces. Now it is `d` for previous and `g` for next. Both keys are next to `f` which I found quite handy to use.

jfclinton, 2015-02-20:
works really efficiently, easy to use with d /g as hot keys. nice

gempa-jabe, 2015-02-20:
Thanks for the feedback, happy to hear that. It is up to you to close the issue if you think it is solved.

closed 2015-02-20 by jfclinton