SeisComP3/seiscomp3 #19: improve display of problem channel info in scolv picker

jfclinton, 2015-02-13:
in scolv picker window when using 3C option, currently if there is no difference in how problems are shown: trace is simply red and channel name is 'valid'. I propose showing that is there is missing metadata, use a different colour than if there is missing waveform data, and instead of labelling the channel a default 1,2,3, use a message as 'no metadata defined'.
(motivation: we migrate 1990s network info, that includes many 1C stations. We continue to want to see 3C if available, its now difficult to easily see if a 'red' bars means missing data or metadata.)

gempa-jabe, 2015-02-20:
I agree.

smallint, 2015-03-04:
Would that fulfill your needs?

[image]

gempa-jabe, 2015-03-04:
Would that fulfill your needs?

[image]

jfclinton, 2015-03-04:
Looks about right! Can we test this at SED?

gempa-jabe, 2015-03-05:
Yes, here it is: 86e1c696137e5b1c873d55630e2f619633a66a10

jfclinton, 2015-04-15:
Jan - looks fine - closing

closed 2015-04-15 by jfclinton