SeisComP3/seiscomp3 #20: use scamp to recompute magnitude for a given origin

marcopovitch, 2015-02-16:
I wanted to recompute all the location and the magnitude for all my events (preferredOriginID).

Combining screloc (with the latest option recently included) and with scamp -O OriginID, and with little change in source code, it is eventually possible.

In the original file src/trunk/apps/processing/scamp/amptool.cpp when scamp is called with -O option the messaging system is disabled :
```
bool AmpTool::validateParameters() {
	if ( !_originID.empty() )
		setMessagingEnabled(false);

	_testMode = commandline().hasOption("test");

	return true;
}
```

To make it recompute the magnitude with scamp -O originID I had to remove the 2 lines below
in order to make scamp use the bus for the new amplitudes computed. Then smag get them and a new set of magnitudes are computed and associated to the originID
```
	if ( !_originID.empty() )
		setMessagingEnabled(false);
```

I wonder if it is possible to rather use that, hopping it does not introduce side effects ?
```
	if ( !_originID.empty() && _testMode )
		setMessagingEnabled(false);
```

gempa-jabe, 2015-02-16:
Shouldn't scamp/scmag trigger automatically if a new origin is sent around by screloc? Actually your fix should work if you keep in mind to run scamp always with `--test` if you don't want it to send messages. I did some work on running all the modules in offline mode and populate an XML file without using the messaging bus at all. I will push the changes as soon it is completed.

marcopovitch, 2015-02-16:
yes,  scamp/scmag trigger automatically. But the point is I wanted scamp to recompute the amplitude because for example at a time I had a wrong Dataless/metadata, and I don't want smag gets its amplitudes from the database.

gempa-jabe, 2015-03-05:
Added, let me know if that works for you.

salichon, 2015-04-14:
Hi !
I tried the scamp patch it worked so far on my side to provide recomputed magnitudes and a newly computed one (added a trial Magnitude plugin) to former events.
Thanks for this it was helpful.

closed 2015-07-02 by gempa-jabe