SeisComP3/seiscomp3 #22: Scart streaming issue when string time is related to non existing SDS  miniseed file

salichon, 2015-03-05:
Using the scart utility to  broadcast miniseed files  into a replay  from a sds directory
for instance:
> seiscomp exec scart --test  -dsvvvE  --list WI_Stat.txt /work/seiscomp/sds

the streaming isn't occuring,  if the miniseed file of the starting day exist for the station mentioned in the station file WIStat.txt. The broadcast of that station  does not operate at all even if the later miniseed files are existing. 

 In another words if the miniseed file of 2011-06-12 is non existing, using this option
> 2011-06-12  23:59:59 ; 2011-06-13 12:05:00;NZ.WI01.*.*
provides No streaming

Using instead this option with an existing  2011-06-13 miniseed file 
> 2011-06-13  00:00:00 ; 2011-06-13 12:05:00;NZ.WI01.*.*
will provide some streaming, even if that miniseed volume is incomplete (and starts later than midnight)

Marc Grumberg suggested to revise the scart python code 
 >In the scart.py  code line 76 :
>	files = glob.glob(dir + "*.%03d" % t[7])
> Perhaps to overcome this behavior, at first sight, one can increment t[7] (jday) until files is not empty ...

Pull requests welcome :) 

cheers
Jerome 

-- 

To be more clear  my first mseed file start 2003.056 and the last one is 2003.115. 
When using scart with -t flag set to "2003-01-01 00:00~2003-12-31 23:59"   (I didn't want to bother me with time) 
it does not produce any file. 
I fact, to have data produced,  I have to set the start date (-t) exactly at 2003-02-25 00:00 (ie. 2003.056) 

I don't know if it is a feature or a bug  but it is a little bit tricky ! 

In the scart.py  code line 76 : 
files = glob.glob(dir + "*.%03d" % t[7]) 

Perhaps to overcome this behavior, at first sight, one can increment t[7] (jday) until files is not empty ... 

Regards 
Marc

gempa-jabe, 2015-03-05:
Check if the patch works. It was actually a bit more work than just increasing the day counter but finally it works across year borders as well.

salichon, 2015-03-09:
Hi !
I made a quick test an a quick run on incomplete data set for which the 
starting day of scart isn't present in the archive
having started :

> 2011-06-12 00:00:00 ; 2011-06-13 02:15:00;NZ.WI01.*.*
where the 12-06 miniseed  isn't present and data starts instead the  
13-06 starts at ~ 2:06

  running:
> seiscomp exec scart --test  -dsE  --list WI_Stat.txt  -vvvv  
> /work/seiscomp/sds

outputs :

> *NZ.WI01.10.HHE 2015-03-09T14:19:50.15335Z 2011-06-13T02:06:32.61Z 
> 2011-06-13T02:06:36.73Z*
> *NZ.WI01.10.HHZ 2015-03-09T14:19:50.153576Z 2011-06-13T02:06:32.61Z 
> 2011-06-13T02:06:36.73Z*

Which is the results expected

Thanks a lot for that quick (but  not that trivial .. ) fix!
Jerome

> Check if the patch works. It was actually a bit more work than just 
> increasing the day counter but finally it works across year borders as 
> well.

-- 
------------------------------------------
Jerome Salichon
Geonet
GNS Science - Te Pu Ao
1 Fairway Drive, PO Box 30368, Lower Hutt, New Zealand
Web: www.gns.cri.nz
Tel. +64 45704 855

Notice: This email and any attachments are confidential.
If received in error please destroy and immediately notify us.
Do not copy or disclose the contents.

gempa-jabe, 2015-03-11:
Thanks for reporting back. According to your message on the mailing list the issue is resolved.

closed 2015-03-11 by gempa-jabe