SeisComP3/seiscomp3 #23: Scautoloc usage in Playback - Epochs and end statement in the XML datatless -

salichon, 2015-03-05:
If the Network, stream, sensorLocation label "<end>" is set as a last statement in the dataless xml files, the station isn't processed in playback mode.

As precised by Jan:
scautoloc reads all stations from inventory where 'now' must match the epoch. This is actually a problem for playbacks of past data. scautoloc should use the stations information of the epoch that matches the pick time but this is currently not the case. Note that only the station itself is important, sensorLocation and stream information are not used at all.

See at least:/seiscomp3/src/trunk/apps/processing/scautoloc/app.cpp
and code lines:

#################################
    DataModel::Network *network = inventory->network(n);

try {
if ( network->end() <= now )

    continue;

}
catch ( ... ) {
for ( size_t s = 0; s < network->stationCount(); ++s ) {
DataModel::Station *station = network->station(s);
try {
if ( station->end() <= now )
continue;
##################################

NB: 
A way to work around that issue is to remove the last <end> statement
To be used with caution though if the dataless volume has a non-unique station history with several <start> and <end>

jsaul, 2018-10-11:
I believe that this has been fixed now. Any feedback will be appreciated so that we can close the issue.

salichon, 2018-10-15:
Hello @jsaul 
thanks for reminding me this, it s very nice to have this fixed!
i ll  proceed to some playbacks tests !

- [ ] use a SC3 inventory  xml with some END date labels on a event occured before that date 

- [ ]  Use Seiscomp3 version Upon October branch compilation or next release.

(see https://github.com/SeisComP3/seiscomp3/pull/197 , https://github.com/SeisComP3/seiscomp3/issues/196) 
and  fix in commit https://github.com/SeisComP3/seiscomp3/commit/136a1d6570d0d9eb90d5d37161a8f1e422f7f1fc 

Thanks @jsaul and @luca-s