SeisComP3/seiscomp3 #24: OSX compilation failes

marcopovitch, 2015-03-16:
Compilation failed on OSX but it is OK on Linux/Ubuntu. This is the compilation error log I've got :

```
[ 99%] Built target _DataModel
Linking CXX shared module ../../../../lib/_Geo.so
Undefined symbols for architecture x86_64:
  "Seiscomp::Geo::GeoFeatureSet::GeoFeatureSet(Seiscomp::Geo::GeoFeatureSet const&)", referenced from:
      _wrap_new_GeoFeatureSet__SWIG_1(_object*, _object*) in GeoPYTHON_wrap.cxx.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [lib/_Geo.so] Error 1
make[1]: *** [src/trunk/libs/swig/CMakeFiles/_Geo.dir/all] Error 2
make: *** [all] Error 2
```

gempa-jabe, 2015-03-16:
Can you post the output when invoked with "-v" to see the actual linker options? Maybe linking against libseiscomp3_core is missing although specified in the CMakeLists.txt file.

marcopovitch, 2015-04-01:
```Linking CXX shared module ../../../../lib/_Geo.so
cd /Users/marc/github/seiscomp3-geoth/build/src/trunk/libs/swig && /usr/local/Cellar/cmake/3.2.1/bin/cmake -E cmake_link_script CMakeFiles/_Geo.dir/link.txt --verbose=1
/usr/bin/c++   -bundle -Wl,-headerpad_max_install_names   -o ../../../../lib/_Geo.so CMakeFiles/_Geo.dir/GeoPYTHON_wrap.cxx.o /usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/libpython2.7.dylib ../../../../lib/libseiscomp3_core.dylib /usr/local/lib/libboost_thread-mt.dylib /usr/local/lib/libboost_iostreams.dylib /usr/local/lib/libboost_filesystem.dylib /usr/local/lib/libboost_regex.dylib /usr/local/lib/libboost_program_options.dylib /usr/local/lib/libboost_thread-mt.dylib /usr/local/lib/libboost_iostreams.dylib /usr/local/lib/libboost_filesystem.dylib /usr/local/lib/libboost_regex.dylib /usr/local/lib/libboost_program_options.dylib /usr/local/lib/libboost_system.dylib ../3rd-party/tau/libtau/libtau_internal.a /usr/lib/libxml2.dylib /usr/lib/libssl.dylib /usr/lib/libcrypto.dylib ../../../../lib/libseiscomp3_config.dylib -liconv ../../../../lib/libmseed.2.6.2.dylib
Undefined symbols for architecture x86_64:
  "Seiscomp::Geo::GeoFeatureSet::GeoFeatureSet(Seiscomp::Geo::GeoFeatureSet const&)", referenced from:
      _wrap_new_GeoFeatureSet__SWIG_1(_object*, _object*) in GeoPYTHON_wrap.cxx.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [lib/_Geo.so] Error 1
make[1]: *** [src/trunk/libs/swig/CMakeFiles/_Geo.dir/all] Error 2
make: *** [all] Error 2```

marcopovitch, 2015-04-01:
I set CMAKE_VERBOSE_MAKEFILE to ON (output is on previous messages)  but I don't kow where to put '-v' option for the linker as suggested.

gempa-jabe, 2015-04-01:
It does not link ../../../../lib/libseiscomp3_core.dylib, I will check this.

gempa-jabe, 2015-04-01:
Sorry, it does, just overlooked it ...

gempa-jabe, 2015-04-01:
Hopefully fixed with https://github.com/SeisComP3/seiscomp3/commit/d14bacd769f0b32abf01e6bbd533daad5be5385c

marcopovitch, 2015-04-01:
Now I have an error saying that:

```
/Users/marc/github/seiscomp3-geoth/src/trunk/libs/swig/GeoPYTHON_wrap.cxx
/Users/marc/github/seiscomp3-geoth/src/trunk/libs/swig/GeoPYTHON_wrap.cxx:7738:48: error: calling a private constructor of class
      'Seiscomp::Geo::GeoFeatureSet'
```

gempa-jabe, 2015-04-01:
Sorry, needed to update the wrappers as well: 53908ff003a5b7903cdcccf2ca711a03fd578ec6.

marcopovitch, 2015-04-01:
ok thank you. It compiles fine !

comoglu, 2015-04-01:
So is it necessary to do anything else then to download source from github
then compile?

marcopovitch, 2015-04-01:
yes :/

you have to install all the dependencies yourself (python, libboost, mysql, etc.)
and make the adjustment in the makefile.

For the dependencies I use "brew" (OSX) to install them
Regards
Marc

comoglu, 2015-04-01:
Yes sure dependencies are crucial :)
all are ready with brew/homebrew
I ll try today.
Regards
Mustafa

comoglu, 2015-04-02:
Hi,
I tried to compile after I downloaded the zip file from git hub. the
previous error message remain. What could be the reason?
Regards

CMake Warning (dev):
   Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run
"cmake
   --help-policy CMP0042" for policy details.  Use the cmake_policy command
to
   set the policy and suppress this warning.

   MACOSX_RPATH is not specified for the following targets:

    mseed
    seiscomp3_client
    seiscomp3_config
    seiscomp3_core
    seiscomp3_datamodel_sm
    seiscomp3_datamodel_vs
    seiscomp3_evplugin
    seiscomp3_mm_preliminaryorigin
    seiscomp3_mplugin
    seiscomp3_qcplugin
    seiscomp3_qt4

Messages during last pass

closed 2015-04-03 by marcopovitch

gilcel, 2015-06-19:
Hi,

I've managed to compile seiscomp3 from the github repository on OS X 10.10 Yosemite with Xcode 6.3.2. The program sc* are installed in /usr/local/bin
First test seems to be OK...but I think not everything is perfect...I need to figure out how to connect now to MySQL database etc.

I'm using Macports with mysql56, qt4-mac
sudo port selfupdate
sudo port install mysql56
sudo port install mysql56-server
sudo port install qt4-mac

It always stopped compiling with mysql:
Add following line at beginning of  seiscomp3-master/src/trunk/CMakeLists.txt
set(CMAKE_LIBRARY_PATH ${CMAKE_LIBRARY_PATH} /opt/local/lib/mysql56/mysql/)

Add following line at beginning to /seiscomp3-master/src/trunk/plugins/database/mysql/CMakeLists.txt

SET(MYSQL_INCLUDE_DIR /opt/local/include/mysql56/mysql/)

Then for spread compilation, there seems to be a problem finding the dylib, it should be in /libspread/
I manually compiled spread in the 3rd-party folder of 'src/trunk/libs/3rd-party/spread/

Change the line seiscomp3-master/src/trunk/libs/seiscomp3/CMakeLists.txt
to read at line 157:

IF(NOT WIN32)
        SET(SPREAD_SOURCE_DIR ${SC3_PACKAGE_SOURCE_DIR}/libs/3rd-party/spread)
        SET(SPREAD_BINARY_DIR ${SC3_PACKAGE_BINARY_DIR}/libs/3rd-party/spread)
        
        SET(SPREAD_LIB_DIR ${SPREAD_BINARY_DIR}/libspread/)

gempa-jabe, 2015-06-19:
> Add following line at beginning of  seiscomp3-master/src/trunk/CMakeLists.txt
> set(CMAKE_LIBRARY_PATH ${CMAKE_LIBRARY_PATH} /opt/local/lib/mysql56/mysql/)

That is something you should change with e.g. ccmake or passing it to cmake. We cannot include any possible mysql installation path. Same for the include dir.

> SET(SPREAD_LIB_DIR ${SPREAD_BINARY_DIR}/libspread/)

Hmm ... I don't have that directory in my build. Why is it libspread in your case? Did anyone else also run into this?

gilcel, 2015-06-19:
I set manually the prefix path of SPREAD 4.3, along with changing SET(SPREAD_LIB_DIR ${SPREAD_BINARY_DIR}/libspread/)
configure --prefix=~/Downloads/seiscomp3-master/src/trunk/libs/3rd-party/spread/; make; make install

I'm not an expert with CMake so I tried some hacks...not ideal, I know, it seemed to work...If you could fix it, it would be nice...

marcopovitch, 2015-06-19:
me neither I don't have this directory ...

gilcel, 2015-06-19:
So with 'homebrew' dependencies, we just can compile seiscomp3 without modifying anything ?
Or are there remaining issues ?

marcopovitch, 2015-06-19:
with brew and dependencies well installed no problem.
You have probably change little thing in `src/trunk/plugins/database/mysql/mysqldatabaseinterface.h` and 
set `#include <mysql/mysql.h>`for OSX. But that's all.

gilcel, 2015-06-19:
Ok thanks, good to know. Will try it on a different machine with 'brew'.
So the same dependencies as the macports I suppose (mysql, libboost, qt4 ?)

Are there other issues with Mac binaries  ?
Do you use it at production level ?

marcopovitch, 2015-06-19:
For the dependencies it is the same as Linux. 
One  issue is that arclink on osx is very slow (I haven't investigate to much but running arclink in a VM is even quicker !). But I mainly use sc3/osx as a client and for testing purpose. 
Our production system is on Linux.

gilcel, 2015-06-19:
Ok, good to know! Thanks for your answers.

gilcel, 2015-06-22:
Ok I just installed Homebrew (instead of Macports) to install the dependencies of the latest seiscomp3 from github / Mac OS X 10.10.3 with Xcode 6.3.2. Here's my first try:

#brew install cmake
#brew install boost
#brew install qt4

To install gfortran:
#brew install gcc
#brew install mysql

It stops compiling here:
[ 52%] Building CXX object src/trunk/libs/seiscomp3/CMakeFiles/seiscomp3_client.dir/communication/spread/spreaddriver.cpp.o
Linking CXX shared library ../../../../lib/libseiscomp3_client.dylib
clang: error: no such file or directory: '../3rd-party/spread/lib/libtspread-core.dylib'
make[2]: *** [lib/libseiscomp3_client.dylib] Error 1
make[1]: *** [src/trunk/libs/seiscomp3/CMakeFiles/seiscomp3_client.dir/all] Error 2
make: *** [all] Error 2

Anyone has the same error ?

ChrisBail, 2015-07-07:
Hi,

I have also some troubles installing Seiscomp3 on mac OSX. I've installed all the dependencies using homebrew. Then I don't really know the worklflow of commands. Fisrt command I've done is:
> cmake CMakeLists.txt

I get the following:
-- The CXX compiler identification is GNU 4.6.0
-- Checking whether CXX compiler has -isysroot
-- Checking whether CXX compiler has -isysroot - yes
-- Checking whether CXX compiler supports OSX deployment target flag
-- Checking whether CXX compiler supports OSX deployment target flag - yes
-- Check for working CXX compiler: /usr/local/bin/g++
-- Check for working CXX compiler: /usr/local/bin/g++ -- broken
CMake Error at /usr/local/Cellar/cmake/3.2.3/share/cmake/Modules/CMakeTestCXXCompiler.cmake:54 (message):
  The C++ compiler "/usr/local/bin/g++" is not able to compile a simple test
  program.

What is the problem?

Thanks

gilcel, 2016-01-08:
Well, I figured out how to compile seiscomp3-jakarta-2015.149 on OS X 10.11 El Capitan with Xcode Command Line Tools and Homebrew.
There are still some issues (with rpath of the spread library not getting called) but it compiles fine, and GUI etc. works and modules get loaded...

- If you don't have XCode.app installed or just want to have the Command Line Tools without the 9GB of XCode.app 7.2 then just do:
```#xcode-select --install```

- Install the GFORTRAN Binaries for El Capitan
https://gcc.gnu.org/wiki/GFortranBinaries

- Install Homebrew's brew command:
```ruby -e "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install)"```

- Install the dependencies as listed by seiscomp install page:
```
#brew install cmake
#brew install boost
#brew install qt4
#brew install flex
#brew install mysql
```

- Download and unzip seiscomp3 in your Downloads folder, then:
```
#cd ~/Downloads/seiscomp3
#make clean
#cmake .
#make -j 2
```
It will compile and at some point it will stop with ***spread lib error***:
Edit file src/trunk/libs/seiscomp3/CMakeLists.txt 

At line 157 or inside IF(NOT WIN32), add the following lines:

```
      	IF(APPLE)
              	SET(SPREAD_LIB_DIR ${SPREAD_BINARY_DIR}/libspread)
         ENDIF(APPLE)
```

So now src/trunk/libs/seiscomp3/CMakeLists.txt looks like (excerpt):
```
      IF(NOT WIN32)
              SET(SPREAD_SOURCE_DIR
      ${SC3_PACKAGE_SOURCE_DIR}/libs/3rd-party/spread)
              SET(SPREAD_BINARY_DIR
      ${SC3_PACKAGE_BINARY_DIR}/libs/3rd-party/spread)
              SET(SPREAD_LIB_DIR ${SPREAD_BINARY_DIR}/lib)
              
              IF(APPLE)
              	SET(SPREAD_LIB_DIR ${SPREAD_BINARY_DIR}/libspread)
            ENDIF(APPLE)
```          

Go back again to ~/Downloads/seiscomp3 and redo a:
```#make -j 2```

It will again stop with a **MySQL error**

```
Building CXX object src/trunk/plugins/database/mysql/CMakeFiles/dbmysql.dir/mysqldatabaseinterface.cpp.o
```

Edit src/trunk/plugins/database/mysql/CMakeLists.txt and add the following lines **before**
INCLUDE_DIRECTORIES(${MYSQL_INCLUDE_DIR}), so it should look like (excerpt):

```
        IF(APPLE)
         SET(MYSQL_INCLUDE_DIR /usr/local/include/mysql/)
        ENDIF(APPLE)

          INCLUDE_DIRECTORIES(${MYSQL_INCLUDE_DIR})
          INCLUDE_DIRECTORIES(${Boost_INCLUDE_DIRS})
```
Go back again to ~/Downloads/seiscomp3 and redo a:

```#make -j 2```

So now it should compile fine...after it has succesfully compiled do a:
```sudo make install```

and everything will be installed in:
/usr/local/
Binaries will be installed in
/usr/local/bin

Just edit your .bash_profile and add the line (recommended by Homebrew)
```export PATH=/usr/local/bin:/usr/local/sbin:$PATH```

Try to source the .bash_profile or open a new Terminal.app window and type
```seiscomp status```

The only issue when calling seiscomp status or scconfig is the rpath error for the 'spread library', I need to figure out how to fix this:

```
error: /usr/local/etc/init/arclink.py: dlopen(/usr/local/lib/python/seiscomp3/_Client.so, 2): Library not loaded: libtspread-core.3.dylib
  Referenced from: /usr/local/lib/python/seiscomp3/_Client.so
  Reason: unsafe use of relative rpath libtspread-core.3.dylib in /usr/local/lib/python/seiscomp3/_Client.so with restricted binary
error: /usr/local/etc/init/trunk.py: dlopen(/usr/local/lib/python/seiscomp3/_Client.so, 2): Library not loaded: libtspread-core.3.dylib
  Referenced from: /usr/local/lib/python/seiscomp3/_Client.so
  Reason: unsafe use of relative rpath libtspread-core.3.dylib in /usr/local/lib/python/seiscomp3/_Client.so with restricted binary
spread               is not running
scmaster             is not running
....
```

Hope this helps someone...