SeisComP3/seiscomp3 #26: boost::none_t problem

marcopovitch, 2015-05-22:
Hi, 

I have a compilation error on OSX using boost 1.58.0 related to boost::none_t  (error message below)
but it compiles fine with Linux boot 1.54.

Do you have any clue ?

[  6%] Building CXX object src/trunk/libs/seiscomp3/CMakeFiles/seiscomp3_core.dir/core/optional.cpp.o
cd /Users/marc/github/seiscomp3-geoth/build/src/trunk/libs/seiscomp3 && /usr/bin/c++   -DSC_ALL_SHARED -DWITH_SVN_REVISION -D_REENTRANT -Dseiscomp3_core_EXPORTS -fPIC -I/usr/local/include -I/Users/marc/github/seiscomp3-geoth/src/trunk/libs -I/Users/marc/github/seiscomp3-geoth/build/src/trunk/libs -I/Users/marc/github/seiscomp3-geoth/src/trunk/../system/libs -I/Users/marc/github/seiscomp3-geoth/build/src/trunk/../system/libs -I/Users/marc/github/seiscomp3-geoth/src/system/libs/3rd-party/libmseed -I/usr/include/libxml2 -I/Users/marc/github/seiscomp3-geoth/src/trunk/libs/seiscomp3/../3rd-party/tau -I/Users/marc/github/seiscomp3-geoth/src/trunk/libs/seiscomp3/../3rd-party/locsat/lib/include -I/Users/marc/github/seiscomp3-geoth/src/trunk/libs/seiscomp3/../3rd-party/locsat/lib/include/db3 -I/Users/marc/github/seiscomp3-geoth/src/trunk/libs/seiscomp3/../3rd-party/locsat/lib/include/css -I/Users/marc/github/seiscomp3-geoth/src/trunk/libs/3rd-party/spread/include    -DSHARED_LIBRARY_SUFFIX="\".dylib\"" -DSHARED_MODULE_SUFFIX="\".so\"" -DSEISCOMP_ROOT=\"/Users/marc/seiscomp3\" -DSEISCOMP_SHARE_DIR=\"share\" -DSEISCOMP_CONFIG_DIR=\"etc/defaults\" -o CMakeFiles/seiscomp3_core.dir/core/optional.cpp.o -c /Users/marc/github/seiscomp3-geoth/src/trunk/libs/seiscomp3/core/optional.cpp

/Users/marc/github/seiscomp3-geoth/src/trunk/libs/seiscomp3/core/optional.cpp:29:30: error: no matching conversion for functional-style
      cast from 'int' to '::boost::none_t'
::boost::none_t const None = ::boost::none_t(0);
                             ^~~~~~~~~~~~~~~~~
/usr/local/include/boost/none_t.hpp:22:7: note: candidate constructor (the implicit copy constructor) not viable: no known conversion
      from 'int' to 'const boost::none_t' for 1st argument
class none_t {};
      ^
/usr/local/include/boost/none_t.hpp:22:7: note: candidate constructor (the implicit default constructor) not viable: requires 0
      arguments, but 1 was provided
1 error generated.
make[2]: *** [src/trunk/libs/seiscomp3/CMakeFiles/seiscomp3_core.dir/core/optional.cpp.o] Error 1
make[1]: *** [src/trunk/libs/seiscomp3/CMakeFiles/seiscomp3_core.dir/all] Error 2
make: *** [all] Error 2

gempa-jabe, 2015-05-22:
Please check f815e8ab8bca818870902a6ef4b2f6536b55611f if that is working for you.

marcopovitch, 2015-05-22:
it works !

closed 2015-05-22 by marcopovitch