SeisComP3/seiscomp3 #39: fdsnws-station inconsistent channel level requests in text format using endafter parameter

ozym, 2015-09-29:
With the fdsnws-station service I'm not sure about this behaviour, if I run the following request multiple times I get different results ... which puzzles me.

`curl 'http://geofon.gfz-potsdam.de/fdsnws/station/1/query?level=channel&format=text&endafter=2015-09-29T00:00:00' | wc -l`

where the endafter parameter is supposed to be the time at which I run the command, to provide currently active channels.

e.g. in quick succession:

`curl 'http://geofon.gfz-potsdam.de/fdsnws/station/1/query?level=channel&format=text&endafter=2015-09-29T00:00:00' | wc -l`
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  446k    0  446k    0     0   117k      0 --:--:--  0:00:03 --:--:--  126k
__3750__

`curl 'http://geofon.gfz-potsdam.de/fdsnws/station/1/query?level=channel&format=text&endafter=2015-09-29T00:00:00' | wc -l`
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  431k    0  431k    0     0  67904      0 --:--:--  0:00:06 --:--:-- 84673
__3624__

`curl 'http://geofon.gfz-potsdam.de/fdsnws/station/1/query?level=channel&format=text&endafter=2015-09-29T00:00:00' | wc -l`
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  446k    0  446k    0     0   116k      0 --:--:--  0:00:03 --:--:--  125k
__3751__

I've replicated this on our geonet service as well. -- Mark

gempa-jabe, 2015-10-05:
Thanks Mark for reporting. We (gempa) fixed that already back in June (https://github.com/SeisComP3/seiscomp3/commit/92a8712aeba2ebd205981f33c5eb053a1d53ebd1) but I am not sure if the Geofon server has been upgraded. This fix will be part of a new release. I tried to replicate it given your URL and it worked. Can you please try again, maybe Geofon upgraded the server.

ozym, 2015-10-06:
Cool, looks like I didn't have the latest source build running and I too no longer see the problem with the geofon service.

closed 2015-10-06 by ozym