SeisComP3/seiscomp3 #42: seiscomp3 on Mac OS X 10.11 libspread rpath issue

gilcel, 2016-01-08:
Hi,

I've succesfully compiled seiscomp3 jakarta 2015.149 on OS X 10.11 El-Capitan with Xcode 7.2.
The GUI starts etc. and the binaries and libs are installed in /usr/local/

For the root CMakeLists.txt I had to set the "rpath" to compile succesfully the libraries in src/trunk//libs/3rd-party/
`set(CMAKE_MACOSX_RPATH 1)`

However I can't figure out how to set "rpath" to the 3rd-party lib: spread 4.3.0.
I tried to built with static libraries only (by editing CMakeLists.txt in src/trunk/libs/seiscomp3/
but no succes.

When I start 'seiscomp status' I get the following error:

```
#seiscomp status
error: /usr/local/etc/init/arclink.py: dlopen(/usr/local/lib/python/seiscomp3/_Client.so, 2): Library not  loaded: libtspread-core.3.dylib
Referenced from: /usr/local/lib/python/seiscomp3/_Client.so
  Reason: unsafe use of relative rpath libtspread-core.3.dylib in /usr/local/lib/python/seiscomp3/_Client.so with restricted binary
error: /usr/local/etc/init/trunk.py: dlopen(/usr/local/lib/python/seiscomp3/_Client.so, 2): Library not loaded: libtspread-core.3.dylib
  Referenced from: /usr/local/lib/python/seiscomp3/_Client.so
  Reason: unsafe use of relative rpath libtspread-core.3.dylib in /usr/local/lib/python/seiscomp3/_Client.so with restricted binary
spread               is not running
scmaster             is not running
....
```
So /usr/local/lib/python/seiscomp3/_Client.so has a problem loading libtspread-core.3.dylib
Even if copy src/trunk/libs/3rd-party/spread/libspread/libspread* to /usr/local/lib/python/seiscomp3/

Here's the list of src/trunk/libs/3rd-party/spread/libspread/
```
 ls -l src/trunk/libs/3rd-party/spread/libspread/libspread*
-rwxr-xr-x  1 seiscomp  staff   62140 Jan  8 15:41 src/trunk/libs/3rd-party/spread/libspread/libspread-core.3.0.0.dylib
-rw-r--r--  1 seiscomp  staff  160920 Jan  8 15:41 src/trunk/libs/3rd-party/spread/libspread/libspread-core.a
lrwxr-xr-x  1 seiscomp  staff      26 Jan  8 15:41 src/trunk/libs/3rd-party/spread/libspread/libspread-core.dylib -> libspread-core.3.0.0.dylib
-rwxr-xr-x  1 seiscomp  staff  154172 Jan  8 15:41 src/trunk/libs/3rd-party/spread/libspread/libspread.3.0.0.dylib
-rw-r--r--  1 seiscomp  staff  375384 Jan  8 15:41 src/trunk/libs/3rd-party/spread/libspread/libspread.a
lrwxr-xr-x  1 seiscomp  staff      21 Jan  8 15:41 src/trunk/libs/3rd-party/spread/libspread/libspread.dylib -> libspread.3.0.0.dylib
```

So the main problem is that the rpath is missing for libtspread-core.3.dylib
Tried also with the latest spread source code 4.4.0 but no luck...

Any idea how to fix this ?

Cheers,

Gilles

gilcel, 2016-01-15:
OK here's the solution so far:

In the latest release of OS X El Capitan 10.11 using the DYLD_LIBRARY_PATH (or DYLD_FALLBACK_LIBRARY_PATH) for security reason is not possible unless System Integrity Protection (SIP) is disabled....
So you need to manually change the library path with the command "install_name_tool -change ...":

```
#seiscomp status
error: /Users/seiscomp/seiscomp3/etc/init/arclink.py: dlopen(/Users/seiscomp/seiscomp3/lib/python/seiscomp3/_Client.so, 2): Library not loaded: libtspread-core.3.dylib
  Referenced from: /Users/seiscomp/seiscomp3/lib/python/seiscomp3/_Client.so
  Reason: unsafe use of relative rpath libtspread-core.3.dylib in /Users/seiscomp/seiscomp3/lib/python/seiscomp3/_Client.so with restricted binary
```
Check with "otool -L" 
```
otool -L /Users/seiscomp/seiscomp3/lib/python/seiscomp3/_Client.so
```
You will see that **libtspread-core.3.dylib** has no library path or @rpath
```
otool -L /Users/sysop/seiscomp3/lib/python/seiscomp3/_Client.so 
/Users/sysop/seiscomp3/lib/python/seiscomp3/_Client.so:
	/System/Library/Frameworks/Python.framework/Versions/2.7/Python (compatibility version 2.7.0, current version 2.7.10)
	/Users/sysop/seiscomp3/lib/libseiscomp3_client.dylib (compatibility version 0.0.0, current version 0.0.0)
	/Users/sysop/seiscomp3/lib/libseiscomp3_core.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/lib/libxml2.2.dylib (compatibility version 10.0.0, current version 10.9.0)
	/usr/local/opt/openssl/lib/libssl.1.0.0.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/openssl/lib/libcrypto.1.0.0.dylib (compatibility version 1.0.0, current version 1.0.0)
	/Users/sysop/seiscomp3/lib/libseiscomp3_config.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/lib/libiconv.2.dylib (compatibility version 7.0.0, current version 7.0.0)
	/Users/sysop/seiscomp3/lib/libmseed.2.6.2.dylib (compatibility version 0.0.0, current version 2.6.2)
	/usr/local/opt/boost/lib/libboost_iostreams.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/opt/boost/lib/libboost_thread-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/opt/boost/lib/libboost_program_options.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/opt/boost/lib/libboost_filesystem.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/opt/boost/lib/libboost_regex.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/opt/boost/lib/libboost_system.dylib (compatibility version 0.0.0, current version 0.0.0)
	libspread/libtspread-core.3.0.0.dylib (compatibility version 3.0.0, current version 3.0.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 120.1.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1226.10.1)
```

To fix we need to change the location of ***libtspread-core.3.0.0.dylib*** of ***_Client.so*** to its path (in /Users/sysop/seiscomp3/libspread/ with install_name_tool -change:

```
install_name_tool -change libtspread-core.3.0.0.dylib /Users/sysop/seiscomp3/libspread/libtspread-core.3.0.0.dylib /Users/sysop/seiscomp3/lib/python/seiscomp3/_Client.so 
```

Checking again with otool -L you'll see that the path of libtspread-core has changed:

```
 otool -L /Users/sysop/seiscomp3/lib/python/seiscomp3/_Client.so 
/Users/sysop/seiscomp3/lib/python/seiscomp3/_Client.so:
	/System/Library/Frameworks/Python.framework/Versions/2.7/Python (compatibility version 2.7.0, current version 2.7.10)
	/Users/sysop/seiscomp3/lib/libseiscomp3_client.dylib (compatibility version 0.0.0, current version 0.0.0)
	/Users/sysop/seiscomp3/lib/libseiscomp3_core.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/lib/libxml2.2.dylib (compatibility version 10.0.0, current version 10.9.0)
	/usr/local/opt/openssl/lib/libssl.1.0.0.dylib (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/openssl/lib/libcrypto.1.0.0.dylib (compatibility version 1.0.0, current version 1.0.0)
	/Users/sysop/seiscomp3/lib/libseiscomp3_config.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/lib/libiconv.2.dylib (compatibility version 7.0.0, current version 7.0.0)
	/Users/sysop/seiscomp3/lib/libmseed.2.6.2.dylib (compatibility version 0.0.0, current version 2.6.2)
	/usr/local/opt/boost/lib/libboost_iostreams.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/opt/boost/lib/libboost_thread-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/opt/boost/lib/libboost_program_options.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/opt/boost/lib/libboost_filesystem.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/opt/boost/lib/libboost_regex.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/opt/boost/lib/libboost_system.dylib (compatibility version 0.0.0, current version 0.0.0)
	/Users/sysop/seiscomp3/libspread/libtspread-core.3.0.0.dylib (compatibility version 3.0.0, current version 3.0.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 120.1.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1226.10.1)
```
In fact you need to change a lot of executables, or plugins...here's a small excerpt
```
# install_name_tool -change libtspread-core.3.dylib /Users/sysop/seiscomp3/libspread/libtspread-core.3.0.0.dylib /Users/sysop/seiscomp3/share/plugins/dbplugin.so
[... cut ...]
# install_name_tool -change libtspread-core.3.dylib /Users/sysop/seiscomp3/libspread/libtspread-core.3.0.0.dylib /Users/sysop/seiscomp3/share/plugins/mlh.so 
etc.

```

But then seiscomp3 will work great on OS X.
Here's a good article:
http://thecourtsofchaos.com/2013/09/16/how-to-copy-and-relink-binaries-on-osx/

I try to find a better solution to compile the spread lib with its library path implemented (it works with gcc -dynamiclib 'install_name /path/to/lib/').
...see also here:
https://github.com/conda/conda-build/issues/279
The best would be to do it with cmake directly, but I'm still learning cmake

Hope this helps

gilcel, 2016-01-25:
I've found a solution to avoid to change the location of the library libtspread-core.3.dylib in serveral binaries:
You need to edit ```src/trunk/libs/3rd-party/spread/libspread/Makefile.in``` and add -install_name '$(SEISCOMP_ROOT)/seiscomp3/libspread/libspread-core.$(LIBFULLVERSION).dylib'  to libspread-core.dylib / libtspread-core.dylib and libspread.dylib.
Since DESTDIR is not defined (empty) in ```src/trunk/libs/3rd-party/spread/libspread/Makefile.in``` I added $(SEISCOMP_ROOT):
```libspread-core.dylib:	$(LIBSP_SHOBJS) $(LIBSPREADUTIL_DIR)/lib/libspread-util.sa
#$(SHLD) -o libspread-core.$(LIBFULLVERSION).dylib $(LIBSP_SHOBJS) $(SHLDFLAGS) $(SHLDCONVERTSTATIC) $(LIBSPREADUTIL_DIR)/lib/libspread-util.sa $(SHLDCONVERTSTATICEND) $(SHLIBS) $(SHLDNAME)libspread-core.$(LIBVERSION).dylib  -current_version $(LIBFULLVERSION) -compatibility_version $(LIBCOMPATVERSION)
$(SHLD) -o libspread-core.$(LIBFULLVERSION).dylib $(LIBSP_SHOBJS) $(SHLDFLAGS) $(SHLDCONVERTSTATIC) $(LIBSPREADUTIL_DIR)/lib/libspread-util.sa $(SHLDCONVERTSTATICEND) $(SHLIBS)  -install_name '$(SEISCOMP_ROOT)/seiscomp3/libspread/libspread-core.$(LIBFULLVERSION).dylib' -current_version $(LIBFULLVERSION) -compatibility_version $(LIBCOMPATVERSION)
$(SOFTLINK) -f libspread-core.$(LIBFULLVERSION).dylib $@```

```libtspread-core.dylib:	$(LIBTSP_SHOBJS) $(LIBSPREADUTIL_DIR)/lib/libspread-util.sa
#$(SHLD) -o libtspread-core.$(LIBFULLVERSION).dylib $(LIBTSP_SHOBJS) $(SHLDFLAGS) $(SHLDCONVERTSTATIC) $(LIBSPREADUTIL_DIR)/lib/libspread-util.sa $(SHLDCONVERTSTATICEND) $(SHLIBS) $(THLIBS) $(SHLDNAME)libtspread-core.$(LIBVERSION).dylib  -current_version $(LIBFULLVERSION) -compatibility_version $(LIBCOMPATVERSION)
$(SHLD) -o libtspread-core.$(LIBFULLVERSION).dylib $(LIBTSP_SHOBJS) $(SHLDFLAGS) $(SHLDCONVERTSTATIC) $(LIBSPREADUTIL_DIR)/lib/libspread-util.sa $(SHLDCONVERTSTATICEND) $(SHLIBS) $(THLIBS) -install_name '$(SEISCOMP_ROOT)/libspread/libtspread-core.$(LIBFULLVERSION).dylib'  -current_version $(LIBFULLVERSION) -compatibility_version $(LIBCOMPATVERSION)
$(SOFTLINK) -f libtspread-core.$(LIBFULLVERSION).dylib $@```

```libspread.dylib:	$(LIBTFL_SHOBJS) $(LIBSPREADUTIL_DIR)/lib/libspread-util.sa
#$(SHLD) -o libspread.$(LIBFULLVERSION).dylib $(LIBTFL_SHOBJS) $(SHLDFLAGS) $(SHLDCONVERTSTATIC) ../stdutil/lib/libstdutil-threaded-release.a $(LIBSPREADUTIL_DIR)/lib/libspread-util.sa $(SHLDCONVERTSTATICEND) $(SHLIBS) $(THLIBS) $(SHLDNAME)libspread.$(LIBVERSION).dylib  -current_version $(LIBFULLVERSION) -compatibility_version $(LIBCOMPATVERSION)
$(SHLD) -o libspread.$(LIBFULLVERSION).dylib $(LIBTFL_SHOBJS) $(SHLDFLAGS) $(SHLDCONVERTSTATIC) ../stdutil/lib/libstdutil-threaded-release.a $(LIBSPREADUTIL_DIR)/lib/libspread-util.sa $(SHLDCONVERTSTATICEND) $(SHLIBS) $(THLIBS) -install_name '$(SEISCOMP_ROOT)/libspread.$(LIBFULLVERSION).dylib' -current_version $(LIBFULLVERSION) -compatibility_version $(LIBCOMPATVERSION)
$(SOFTLINK) -f libspread.$(LIBFULLVERSION).dylib $@```

After recompiling with make no need to use again otool and install_name_tool again !

[Makefile.in.txt](https://github.com/SeisComP3/seiscomp3/files/103405/Makefile.in.txt)

gempa-jabe, 2016-02-03:
Since we cannot maintain an OSX version would you mind to create a branch for the changes that have to be applied? Would you volunteer to maintain that branch?

gilcel, 2016-02-03:
Hello,

I just finished the porting of seiscomp3-jakarta-2015.149 to OS X 10.11 El Capitan with all the modules today! Also the NonLinLoc works now, since I've found a solution to fmemopen & open_memstream functions not working in the nll plug-in.
This will work out of the box (means compiling and no more hacking needed with libspread etc.)
Please see the attached screenshot.

I will try to make a new seiscomp-os-x branch on github....need just to check how github works before screwing everything up :-) Any advice here ?

[image]

gempa-jabe, 2016-02-03:
Looks good :)

> I will try to make a new seiscomp-os-x branch on github....need just to check how github works before screwing everything up :-) Any advice here ?

I have to check that. Either fork the repository and create a pull request into our osx branch (which does not exist yet) or I can invite you as collaborator. Not sure yet how to assign access rights to several branches. Forking might be the easiest solution.

gilcel, 2016-02-03:
Well it works with the latest "boost 1.60" (you'll get a lot of BOOST_JOIN errors, with QT4 and libboost 1.60 (I hadn't these issues with libboost 1.59). So be prepared ;-)
The solution:
You'll need to add #ifndef Q_MOC_RUN for a lot of guit-qt4 source code)

gempa-jabe, 2016-02-03:
> You'll need to add #ifndef Q_MOC_RUN for a lot of guit-qt4 source code)

This has been fixed already. I am about to sync master at the moment.

gempa-jabe, 2016-02-03:
I have pushed the changes. If possible, try again to build your version with current master.

gilcel, 2016-02-03:
Any issue by using the latest spread v4.40 which has higher throughput, lower latency in LANs ?

gempa-jabe, 2016-02-03:
I haven't tried but should work. If everything compiles I don't see any issues unless they broke something which I doubt. Give it a try ...

gilcel, 2016-02-03:
Ok, will try...also for Locator NLL the OS X version needs these libs / files for fmemopen / open_memstream from here:
https://github.com/materialsvirtuallab/pyhull/tree/master/src/fmemopen

Any issues to use it in seiscomp3 with the Apache 2 license ?

gempa-jabe, 2016-02-03:
Actually the Apache 2 license shouldn't be an issue as long as we provide clear attribution to The Apache Software Foundation for any distributions that include Apache software, whatever that means in practice. Apache 2 is not compatible with GPL 2 but GPL 3. I am not sure about the Spread license and Apache 2. For the time being I think it is OK. I am not a lawyer ;)

gempa-jabe, 2016-03-02:
I close this issue and we can reopen it as required.

closed 2016-03-02 by gempa-jabe