SeisComP3/seiscomp3 #44: [fdsnws] ResponsePAZ string required at start of publicID

jordi-domingo, 2016-01-25:
The response elements whose publicID do not start with "ResponsePAZ", "ResponseFIR", etc. do not appear in the output of fdsnws.

Is that an intended behaviour? Isn't it supposed that all publicIDs should start with "smi:organization.string/" (quakeml standard)?

gempa-jabe, 2016-01-25:
I don't get this. You mean if a publicID for a response does not start with e.g "ResponsePAZ" the response it is being ignored in fdsnws? What format is your output, SC3 XML or FDSNXML?

> Isn't it supposed that all publicIDs should start with "smi:organization.string/" (quakeml standard)?

There is no QuakeML station dialect. The output of fdsnws is either SC3 XML or FDSNXML. Neither of those formats enforces an ID schema.

jordi-domingo, 2016-01-25:
Let me explain our case...

At KNMI we are running the SC3 FDSNws. What happened was that for some stations the fdsnws returned the ResponsePAZ information and for others not.

Comparing them we discovered that the only difference was that the format of the publicID for the <responsePAZ> elements in the inventory XML files was diferent. When the publicID started with "ResponsePAZ#", it worked correctly, but when the publicID started with "publicID#", then this ResponsePAZ didn't appear in the fdsnws.

So, we changed the <responsePAZ> publicIDs in the inventory XML files, updated the inventory (ensuring that in the database the publicIDs were also updated) and then it worked!

It also happened to Stefan Heimers from ETHZ/SED, as he explained in the EIDA maintenance mailing list.

> I don't get this. You mean if a publicID for a response does not start with e.g "ResponsePAZ" the response it is being ignored in fdsnws? What format is your output, SC3 XML or FDSNXML?

If I remember well, it happened for both formats.

> There is no QuakeML station dialect. The output of fdsnws is either SC3 XML or FDSNXML. Neither of those formats enforces an ID schema.

True, forget that.

I repeat, I don't know if it is a bug or an intended behaviour. Just that you know that it is dependent on the format of the publicID created for response elements.

gempa-jabe, 2016-01-25:
> I repeat, I don't know if it is a bug or an intended behaviour. Just that you know that it is dependent on the format of the publicID created for response elements.

If this is the case then it is a bug. Thanks for letting us know.

jordi-domingo, 2016-01-25:
You're welcome

gempa-jabe, 2016-01-26:
It should be fixed with 4ff2a87b04579773467fd9e35a2ed4c5b352e5b1. If you can test it let us know if it works for you.

sheimers, 2016-01-26:
Thanks for the quick fix. It works great for us.

We are aware that sc3 inventory xml is different from quakeML, but it makes life easier when we can use the same publicIDs in both Seiscomp3 and in quakeML files we create from data in the Sc3 database.

gempa-jabe, 2016-01-26:
> it makes life easier when we can use the same publicIDs in both Seiscomp3 and in quakeML files we create from data in the Sc3 database.

You can. fdsnws does not change your publicIDs but it does not convert internal publicIDs to any schema.

closed 2016-01-26 by gempa-jabe