SeisComP3/seiscomp3 #56: Useless constraint ?

fabienengels, 2016-03-29:
Hi,

By looking to improve database performance (using PostGreSQL), I notive some constraints start a long time to be checked (when the database starts to be huge). By removing them, it greatly improve delete performance many others :

```sql
ALTER TABLE amplitude DROP CONSTRAINT amplitude__parent_oid_fkey1;
ALTER TABLE pick DROP CONSTRAINT pick__parent_oid_fkey;
ALTER TABLE event DROP CONSTRAINT event__parent_oid_fkey;
ALTER TABLE origin DROP CONSTRAINT origin__parent_oid_fkey;
ALTER TABLE outage DROP CONSTRAINT outage__parent_oid_fkey;
ALTER TABLE responsefir DROP CONSTRAINT responsefir__parent_oid_fkey;
```

I would like to discuss if they are manadatory as in every case, _parent_oid has only one value :

```sql
select count(distinct _parent_oid) from amplitude;
 count 
-------
     1

select count(distinct _parent_oid) from event;
 count 
-------
     1

select count(distinct _parent_oid) from origin;
 count 
-------
     1

select count(distinct _parent_oid) from outage;
 count 
-------
     1

select count(distinct _parent_oid) from responsefir;
 count 
-------
     1

select count(distinct _parent_oid) from pick;
 count 
-------
     1
```

Is _parent_oid used ? and is it risky to remove these constraint ? it really helps us to improve merge time on our database.

gempa-jabe, 2016-03-29:
In case of top level objects as identified in your list it would not be important. Currently in SC3 we only use *one* EventParameters instance but in theory multiple instances could be used. Anyway, the code requires the attribute *_parent_oid* but not the constraint. Almost any constraint is enforced on client side so you can remove it on server side.

fabienengels, 2016-03-31:
Thank you for the answer. Don't you think it could be interesting to push that change in the mainstream distribution ?

gempa-jabe, 2016-03-31:
Actually those were added as foreign keys to the table. It would be interesting to know what is causing the delay. Is it the constraint itself or the delete cascade? I suppose the latter because anytime an object is being removed PostgreSQL must check whether any of the tables links to that particular id. I think we can drop them.

gempa-jabe, 2016-03-31:
I am wondering why e.g. amplitude__oid_fkey does not introduce a delay or does it? That one is a foreign key to the object table, too.

gempa-jabe, 2016-03-31:
See 885d9c3a900731b5d402c1c428b05652e34b98df.

fabienengels, 2016-03-31:
To spot this bottleneck, I simply run :

```sql
explain analyze delete from publicobject where _oid = <random_oid>;
```

and I noticed PostGreSQL spent quite a while to check these constraints (even with indexes) for the given tables, it was about 80% of the time (sorry I didn't keep my result and since our database was cleaned up). For a small database isn't an issue but for the large ones, it was especially during merge of events (I guess SC delete some objects during the process). Commit 885d9c3a900731b5d402c1c428b05652e34b98df is a good thing as many SC databases are in production for many years now. :)

closed 2016-03-31 by fabienengels