SeisComP3/seiscomp3 #59: symlink to seiscomp executable not resolved to absolute path

grapagnani, 2016-04-26:
Issue encountered in: **seiscomp3-jakarta-2016.062.01-debian7-x86_64**

After decompressing to `$HOME` directory. If you create a symlink to the `seiscomp` executable, eg.

`seismo@seislab:~$ ls -l $HOME/bin/seiscomp`
`lrwxrwxrwx 1 seismo seismo 25 Apr 26 05:39 bin/seiscomp -> ../seiscomp3/bin/seiscomp`

If you then run: 
`seismo@seislab:~$ seiscomp status`

You get the error:

`Traceback (most recent call last):`
`  File "bin/seiscomp", line 996, in <module>`
`    sys.exit(system(sys.argv))`
`  File "bin/seiscomp", line 38, in system`
`    proc = subprocess.Popen(args, shell=False, env=os.environ)`
`  File "/usr/lib/python2.7/subprocess.py", line 679, in __init__`
`    errread, errwrite)`
`  File "/usr/lib/python2.7/subprocess.py", line 1259, in _execute_child`
`    raise child_exception`
`OSError: [Errno 2] No such file or directory`

The issue is in the `seiscomp3/bin/seiscomp` executable, line 934:
932:# Resolve symlinks to files (if any)
933:if os.path.islink(sys.argv[0]):
934:  sys.argv[0] = os.readlink(sys.argv[0])
935:
936:# Guess SEISCOMP_ROOT from path of called script, directory links are not

`os.readlink()` does not resolve to the absolute path, it merely returns the file pointed by the executable, in my example: `../seiscomp3/bin/seiscomp`

Reverting to the use of `os.path.realpath()` solves the issue:
934:  sys.argv[0] = os.path.realpath(sys.argv[0])

gempa-jabe, 2016-04-26:
Valid concern and thanks for the suggestion of a fix.

gempa-jabe, 2016-05-11:
Fixed with aa1d12ad983fc491559b8d4f8b6eb772581e28ab, does that resolve the issue on your side?

grapagnani, 2016-05-13:
Hello Jan. Yes, it works correctly now. Thank you very much.

closed 2016-05-13 by gempa-jabe