SeisComP3/seiscomp3 #62: scimport does not import comments

yannikbehr, 2016-05-09:
It seems like scimport is ignoring comments. I am importing Magnitudes that have additional information attached to as comments. The comments, however, do not show up in the database and scimport's log file shows lines like the following:

```
2016/05/09 00:53:55 [debug/ScImport] Filter for class: Comment not available
```

gempa-jabe, 2016-05-11:
That is right, scimport does not import comments. Adding a filter in the code is easy, the difficulty is to filter comments for various parent types. We could either import all or no comments. So the best would be to add an option to allow import of comments or not.

yannikbehr, 2016-05-11:
Would it cause problems if a comment is imported but not its parent?

closed 2018-11-01 by gempa-jabe