SeisComP3/seiscomp3 #66: scevtlog does not log FocalMechanism to XML

gempa-jabe, 2016-05-25:
Issue reported by @salichon, track it here to not forget. scevtlog does neither export the focal mechanism nor the origin of preferred magnitude if it is not part of the preferred origin.

gempa-jabe, 2016-05-31:
I fixed that with 14db7a4e39b01936279c39b526d5d860c0e7b1f8. I have been running scevtlog for three days, gathered a lot of XML and all events with a preferred focal mechanism do contain a FocalMechanism including the derived origin with magnitudes.

quiffman, 2016-05-31:
Thanks @gempa-jabe!

I can give this a try with @salichon and report back if you like.

salichon, 2016-05-31:
Hey Jan
Thanks for that quick update!
I ll run that asap
Cheers
Jerome

salichon, 2016-06-06:
@gempa-jabe, @quiffman  We made some tests: the xml files are logged okay comprising the focalmechanism solutions  and the Derived origins. I haven't seen any issue on my side.
Thanks for the fix.

gempa-jabe, 2016-06-10:
Thank you for testing. I close this issue then.

closed 2016-06-10 by gempa-jabe