SeisComP3/seiscomp3 #67: Database schema version shouldn't appear in publicID

fabienengels, 2016-06-10:
Hi,

We found an issue on how publicID are generated in QuakeML issued by fdsn-event. The database schema version appears in the smi, ex :
publicID="smi:scs/0.7/eost2016ljdn"

Database schema version shouldn't be used in publicID as they are not supposed to change over time.

fabienengels, 2016-07-01:
Don't know if anyone read this issue, should I send a email to the mailing list ? Thank you in advance

gempa-stephan, 2016-07-04:
The SC3 QuakeML exporter used by the fdsnws module uses the fixed prefix `smi:scs/0.7/`. It will not change when updating to 2016.161 respectively database version 0.8.

I guess instead of a fixed prefix you want to see DNS name of the organization here. Therefore the exporter interface needs to be changed and a configuration parameter should be added.