SeisComP3/seiscomp3 #72: FDSNws error when simultaneous large event queries

jordi-domingo, 2016-08-26:
Dear all,

We have problems with our FDSNws service when there are simultaneous (2,3,4...) fdsnws-event medium-large queries (i.e.: 30 events). The module give the following errors (lots of lines):

```
[error/MYSQL] beginQuery: nested queries are not supported
[error/DatabaseArchive] starting query 'select * from EventDescription where EventDescription._parent_oid='164405'' failed
[error/MYSQL] beginQuery: nested queries are not supported
[error/DatabaseArchive] starting query 'select * from OriginReference where OriginReference._parent_oid='164405'' failed
(...)
```
After multiple errors messages, the module returns a incomplete result or event shuts down with the following message:

`[warning/DatabaseArchive] Unable to read schema version from database, assuming v0.0`

I searched in the code and the error seems to be produced here...

```
src/trunk/plugins/database/mysql/mysqldatabaseinterface.cpp

bool MySQLDatabase::beginQuery(const char* q) {
	if ( _result ) {
		SEISCOMP_ERROR("beginQuery: nested queries are not supported");
		//SEISCOMP_DEBUG("last successfull query: %s", _lastQuery.c_str());
		return false;
	}
(...)
```

...however, I don't understand the reason of the fail. Is it a SeisComP3 problem or it comes from the MySQL server?

I tried to reproduce it in other systems and it happens as well, but for some of these systems I had to do more and larger queries (i.e.: 6 concurrent queries of ~6000 events).

Do you have any clue why this happens?

Many thanks

gempa-jabe, 2016-08-26:
Thank you for the report. Indeed that is a problem of FDSNWS, not SeisComP3 in general and also not oyur MYSQL server. The issue exhibits if another query is started through the same database connection while another query is still running, e.g. a result set is still being fetched. We will take a look.

jordi-domingo, 2016-08-26:
Ok. Thanks!

commit 55b9bb9d (2016-08-29): [fdsnws] Fix DB error on simultaneous event queries, fixes #72
closed 2016-08-29 by gempa-jabe