SeisComP3/seiscomp3 #75: scolv events tab pop-up event info window

salichon, 2016-09-22:
An option to dismiss the yellow pop-up window in the events tab would be useful to set:
While selecting or inspecting events this pop-up window is not convenient since appearing each time the cursor is sat on an event.

[image]

gempa-jabe, 2017-12-04:
Rather than adding an additional configuration option I would prefer to show the tooltip only if the ID column is being hovered.

salichon, 2017-12-05:
Great
 i saw this updated in the 2017.334 patch 1 
[https://github.com/SeisComP3/seiscomp3/blob/release/jakarta/CHANGELOG.md](url)
thanks  for this update 
i ll test it  with local compilation and close the ticket ;)  
Cheers
Jerome

salichon, 2018-04-18:
@gempa-jabe  sorry for the late confirmation.
All good tool tip shown on  the ID only.
nice thanks

closed 2018-04-18 by salichon