SeisComP3/seiscomp3 #76: scalert

duperray, 2016-09-27:
Hi,
Got a python exception in scalert, about unknown object System.Environment.
A quick and dirty fix, was to comment:
if self._eventScript:
        self._eventScript = seiscomp3.System.Environment.Instance().absolutePath(self._eventScript)

Cheers,

Pierre

gempa-jabe, 2016-09-27:
I have tried a seiscomp3.System.Environment call in scalert and it works for me. Can you post  the exception? Does your scalert version import seiscomp3.System?

duperray, 2016-09-29:
no my scalert version tagged 2016.161 does not import seiscomp3.System, the exception was about missing Environment or System (sorry, I really have no time yet to test this, I should have wrote the exception)
But I think you hit the nail on the head.
sorry for the late answer

gempa-jabe, 2016-09-29:
You are right. That has been fixed already by Monsieur Grunberg ;), see 98f4d068a87c6d66c35fac4daaf36f242e159a91.

closed 2016-09-29 by gempa-jabe