SeisComP3/seiscomp3 #78: sqlite3 -init

FMassin, 2016-10-05:
`sqlite3 -batch -init $SEISCOMP_ROOT/share/db/sqlite3.sql test_db.sqlite`

works with `sqlite -version`
3.8.7.1 2014-10-29 13:59:56 3b7b72c4685aa5cf5e675c2c47ebec10d9704221

but doesn't finish with `sqlite3 -version`
3.11.0 2016-02-15 17:29:24 3d862f207e3adc00f78066799ac5a8c282430a5f

I will investigate more, but a fix already exists maybe ?

gempa-jabe, 2016-10-05:
Any error? I don't know of a fix.

FMassin, 2016-10-06:
No error that I can see, sorry, just 
> no such table:
when `test_db.sqlite` doesn't exit. After that the command seems to be _frozen_.

Turning `.echo on` shows that the into script  runs fine but it doesn't exit at the end, I thing we just need to find the way to stop  it in a proper manner ...

gempa-jabe, 2016-10-06:
Thanks. `no such table` sounds like it hangs at `drop table foo`. If have already a local fix that replaces that with `drop table if exists foo`. Can you try that to replace `drop table` with `drop table if exists` and check again?

FMassin, 2016-10-06:
Adding `.exit` : 
`sqlite3 -batch -init $SEISCOMP_ROOT/share/db/sqlite3.sql test_db.sqlite  .exit `
seems to work.

Problem fixed I guess ?

closed 2016-10-06 by FMassin

FMassin, 2016-10-07:
and thank you for the patch 👍