SeisComP3/seiscomp3 #80: scenvelope

comoglu, 2016-10-15:
I just compiled the latest git version, everything seems ok except scenvelope. 
database version incompatibility.

2016/10/14 22:14:19 [error/DataModel] Archive version 0.8 too high: Envelope skipped
2016/10/14 22:14:19 [error/Communication] failed to serialize archive
2016/10/14 22:14:19 [error/DataModel] Archive version 0.8 too high: Envelope skipped
2016/10/14 22:14:19 [error/Communication] failed to serialize archive
2016/10/14 22:14:19 [error/DataModel] Archive version 0.8 too high: Envelope skipped
2016/10/14 22:14:19 [error/Communication] failed to serialize archive
2016/10/14 22:14:19 [error/DataModel] Archive version 0.8 too high: Envelope skipped
2016/10/14 22:14:19 [error/Communication] failed to serialize archive
2016/10/14 22:14:19 [error/DataModel] Archive version 0.8 too high: Envelope skipped
2016/10/14 22:14:19 [error/Communication] failed to serialize archive
2016/10/14 22:14:19 [error/DataModel] Archive version 0.8 too high: Envelope skipped
2016/10/14 22:14:19 [error/Communication] failed to serialize archive
2016/10/14 22:14:19 [error/DataModel] Archive version 0.8 too high: Envelope skipped
2016/10/14 22:14:19 [error/Communication] failed to serialize archive
2016/10/14 22:14:19 [error/DataModel] Archive version 0.8 too high: Envelope skipped
2016/10/14 22:14:19 [error/Communication] failed to serialize archive
2016/10/14 22:14:19 [error/DataModel] Archive version 0.8 too high: Envelope skipped

gempa-jabe, 2016-10-15:
Thanks Mustafa: The Git is still work in progress and we will fix that issue.

gempa-jabe, 2016-10-15:
Try the latest commit 0ebd49d2c8ab5bdf87dd03679919bcf9805d4f04 which should fix it.

comoglu, 2016-10-15:
Could you remind me How can I apply the fixes? long time has passed since I have applied a patch. 
Thanks so much.

gempa-jabe, 2016-10-15:
Just pull the lastest git version, thats all.

comoglu, 2016-10-15:
then compile again, isnt it. 
OK I ll do that. 
Thanks Jan.

comoglu, 2016-10-15:
Yep it works now. Thanks

closed 2016-10-24 by gempa-jabe