SeisComP3/seiscomp3 #82: IASP91 Pn Sn  arrival times interpolation tables  issues at short distances  < 1degree

salichon, 2016-10-27:
For distance range <1 degree Picking Pn or Sn phases leads to  irrelevant (large) time residuals in SCOLV
As per a @jsaul e-mail: 

> > Is there a small issue using Pn picks with LOCSAT and the default
> >  velocity models (Iasp91/Tab) at short range ?
yes there is, thanks for reporting!  ... 
The discrepancy between the LocSAT result and what you saw in the picker indicated that it had to have something to do with the tables. Pn cannot be observed at very near distances, at least not for the continental iasp91 model. Geometrically, Pn is observed only at distances beyond the critical distance, which is abound 0.7° or 77 km for iasp91. Now the interpolation of the LocSAT tables comes into play. The default iasp91.Pn table has a distance sampling interval of 0.5 degrees, with the first valid sample at 1.0°! The fact that LocSAT considers Pn picks at distances less than that is most likely a bug in the LocSAT version used in SC3.
The good news: Since the Pn slowness is practically constant (namely 13.754 s/°), we can easily extrapolate the traveltimes down to zero distance in order to allow LocSAT to properly interpolate Pn times at distances of less that 1°.
The attached iasp91.Pn table fixes the problem. I added another depth at 35 km to avoid possible numerical issues around iasp91 Moho depth. Just copy the attached file to where the current iasp91.Pn table is on your system. For me this has solved the issue out of the box and I hope it helps at your end, too.
Cheers
Joachim 

I think the iasp91.Sn should  the same issue, see  
[ seiscomp3/src/trunk/libs/3rd-party/locsat/data/iasp91.Sn ](url)
[ seiscomp3/src/trunk/libs/3rd-party/locsat/data/iasp91.Pn ](url)

Here follows Joachim s update Pn files  
[iasp91.Pn.txt](https://github.com/SeisComP3/seiscomp3/files/554736/iasp91.Pn.txt)

Scolv with close Pn
[image]

gempa-jabe, 2016-10-27:
Is the objective to also fix Sn and to include the fixed phase files into the main distribution?

salichon, 2016-10-27:
Hi @gempa-jabe, Thanks Yes, this would be great.  Also it would be more consistent to have the same interpolation for both files iasp91 Pn and Sn. What do you reckon @jsaul ?