SeisComP3/seiscomp3 #92: scvsmag crashes if origin has no OriginQuality.azimuthalGap

FMassin, 2017-02-01:
Need to fix the dependence to OriginQuality.azimuthalGap in https://github.com/SeisComP3/seiscomp3/blob/master/src/sed/apps/scvsmag/scvsmag.cpp#L674

If there is no P-wave arrival, there is no point to calculate Mvs. 

Feel free to jump in @yannikbehr ...

yannikbehr, 2017-02-07:
@FMassin I guess this is for origins coming from finder? Does an origin from finder have any other references to the raw data it is based on?

FMassin, 2017-02-07:
Yes, this is a question for @gempa-jabe ...

gempa-jabe, 2017-02-07:
No, they don't have arrivals. Just an origin with some quality parameters. We can't declare picks I think.

gempa-jabe, 2017-02-07:
Actually, since scfinder already maintains a buffer of envelopes it should be possible to compute VS magnitudes inside scfinder already. But that is another topic that I don't want to open here and does not correlate to the crash of scvsmag that has been fixed already.

FMassin, 2017-02-07:
Very happy to read this @gempa-jabe I am thinking about since the first time I've heard of FinDer. The issue is open !

yannikbehr, 2017-02-07:
Interesting thought. In principle VS doesn't need a pick, only a location and a stream with corresponding envelope values. However, the potential strength of VS is that it should work with less waveform data than finder. If a fast location algorithm would provide a rough first location, VS may be able to give a equally rough but fast magnitude estimate before finder does. Not sure what can be gained by tying VS magnitudes to finder locations.

gempa-jabe, 2017-02-07:
From my technical understanding the gain could be savings of unnecessary messaging transports and duplicate waveform acquisition. Actually VS receives envelopes and computes magnitudes based on those. scfinder does compute those envelopes, too, buffers them and tries to find a location with the Finder lib. The wrapper (scfinder) converts the detection to an Origin and sends it. This is what VS would like to receive as fast location. Before sending the Origin, scfinder could compute VS magnitudes based on the envelopes it still has stored in memory as scvsmag does. This was just out of my mind and duplicating VS logic in scfinder would tie them strongly together which is probably not desired. And it would break the modularity of SC3.

Again, this is independent of the issue reported here. We should continue the discussion in the new issue.

closed 2017-03-03 by gempa-jabe