SeisComP3/seiscomp3 #95: SeisComP3 compilation errors

dryuanye, 2017-02-07:
Dear colleagues,
I try to compile the SeisComP3 in my new CentOS. The following errors occured:
[ 37%] Building CXX object src/trunk/libs/seiscomp3/CMakeFiles/seiscomp3_client.dir/qc/qcprocessor.cpp.o
Linking CXX shared library ../../../../lib/libseiscomp3_client.so
/usr/bin/ld: /home/sysop/Downloads/seiscomp3-master/src/trunk/libs/3rd-party/spread/lib/libtspread-core.a(sp.to): relocation R_X86_64_32 against `.bss' can not be used when making a shared object; recompile with -fPIC
/home/sysop/Downloads/seiscomp3-master/src/trunk/libs/3rd-party/spread/lib/libtspread-core.a: could not read symbols: Bad value
collect2: error: ld returned 1 exit status
make[2]: *** [lib/libseiscomp3_client.so] Error 1
make[1]: *** [src/trunk/libs/seiscomp3/CMakeFiles/seiscomp3_client.dir/all] Error 2
make: *** [all] Error 2

Do you have any idea about solving the problem? Thank you very much

gempa-jabe, 2017-02-07:
What CentOS release? We build on 5, 6 and 7 and it works. Is it a clean build? You can try to delete src/trunk/libs/3rd-party/spread/lib/* to build again.

dryuanye, 2017-02-07:
ok , we are using a X86 64 machine, we add ADD_DEFINITIONS(-fPIC) at the end of Cmakelist.txt. all works . Thank you very much

closed 2017-02-07 by dryuanye