SeisComP3/seiscomp3 #98: which module should compute envelopes?

ccauzzi, 2017-02-08:
I followed the spin-off discussion generated by issue #92. I believe and concur with Yannik that we should avoid creating any direct  dependence of VS on FinDer. This would be a conceptual mistake as VS has the potential to be faster than FinDer if informed by the location generated by a faster algorithm. Rather I would support the idea of an independent envelope module that feeds all downstream modules requiring envelope estimates.

gempa-jabe, 2017-02-08:
Carlo, that hasn't been suggested to my understanding. VS should not depend on FinDer but the idea was that scfinder can compute magnitudes itself like scvsmag does, probably through a VS library to share common code. That was just an idea to reduce bandwidth as envelope information is already in scfinder. I personally don't prefer the one or the other. Having an envelope module such as scenvelope is fine as long as the computed envelopes are generic enough to be used by many modules. As I understood this is not necessarily the case, that's why the development of libeewamps has been started. And please lets not have Finder discussions in the public SC3 issues as this module is not yet available to the community.

ccauzzi, 2017-02-08:
Fine.

closed 2017-02-08 by ccauzzi