SeisComP3/seiscomp3 #99: The 'publicID' and 'response' time format in generated SC3 XML

pnoker, 2017-02-16:
As shown in the following code,he is through the `./seiscomp exec import_inv dlsv CI_BEL.dataless` command to generate SC3 XML file, I only know a part of the meaning, please help me to find out what the time format is,thanks!
`<sensor publicID="Sensor#20170216055922.099488.72" name="BEL.2001.074.HE" response="ResponsePAZ#20170216055922.099682.73">`
`20170216` -> ` 2017-02-16`,`055922.099488.72` -> ??. The other I don't understand,Please help me!

gempa-jabe, 2017-02-16:
The *publicID* and *response* attribute are IDs, do not interpret them. The IDs have been created incorporating timestamps but it is just an ID. The sensor.publicID identifies the sensor. E.g. stream.sensor refers to that ID. sensor.response refers to the response of the sensor, or more precisely to its publicID:

sensor.publicID = ID of the sensor
sensor.response -> response[fir|paz|polynomial].publicID

pnoker, 2017-02-20:
Thank you very much!

pnoker, 2017-02-20:
Dear author:
I am a Seismological Bureau staff, your software I have been in use, and feel very good. Could I separate the use of separate modules, such as: I would like to separatethe `dlsv2inv` and `inv2dlsv`, used alone?

gempa-jabe, 2017-03-03:
What do you mean with "separate them"? You can call them alone, yes.

closed 2017-12-04 by gempa-jabe