SeisComP3/seiscomp3 #100: fdsnws - small errors in application.wadl and URL builders

jordi-domingo, 2017-02-24:
There are small typos or errors in fdsnws application.wadl files and URL builders pages:

- src/trunk/apps/fdsnws/share/dataselect.wadl
`	<resources base="http://localhost:8080/fdsnws/1/dataselect">`
should be
`	<resources base="http://localhost:8080/fdsnws/dataselect/1">`

- src/trunk/apps/fdsnws/share/event.wadl.wadl
`	<resources base="http://localhost:8080/fdsnws/1/event">`
should be
`	<resources base="http://localhost:8080/fdsnws/event/1">`

- src/trunk/apps/fdsnws/share/station.wadl
`	<resources base="http://localhost:8080/fdsnws/1/station">`
should be
`	<resources base="http://localhost:8080/fdsnws/station/1">`

- src/trunk/apps/fdsnws/share/event-builder.html
  - `maxmagnitude` is missing, instead `minmagnitude` is duplicated (both label and input)
  - the `name` of Include Arrivals input is `includeallmagnitudes`, instead of `includearrivals`
  - Format option QuakeML-RT has `value` empty instead of `qml-rt`
  - Additional, non standard parameters, are missing (done in purpose?): `includepicks`, `includecomments`, `formatted`

- src/trunk/apps/fdsnws/share/station-builder.html
  - Additional, non standard parameters, are missing (done in purpose?): `formatted`

gempa-stephan, 2017-02-24:
Thanks for reporting these errors. The non standard parameters should be listed since the service implements them.

Would you mind to implement those changes and then file a merge request?

jordi-domingo, 2017-02-24:
Ok, no problem.

jordi-domingo, 2017-02-24:
Btw, while adding the non standard parameters, I've detected that:
- `includepicks` is by default true, but the docs say it is by default false. Shall I change event.wadl, event.html and fdsnws.rst; or do you want to change the code to be false by default?
- `includecomments=true` seems to not work (see log below), although it is supposed to be true by default. Shall I create a separate issue for that?

`2017/02/24 13:33:46 [error/log] FDSNEvent instance has no attribute '_loadComment' ['  File "/home/sysop/.local/lib64/python2.7/site-packages/Twisted-15.3.0-py2.7-linux-x86_64.egg/twisted/python/threadpool.py", line 196, in _worker\n    result = context.call(ctx, function, *args, **kwargs)\n', '  File "/home/sysop/.local/lib64/python2.7/site-packages/Twisted-15.3.0-py2.7-linux-x86_64.egg/twisted/python/context.py", line 118, in callWithContext\n    return self.currentContext().callWithContext(ctx, func, *args, **kw)\n', '  File "/home/sysop/.local/lib64/python2.7/site-packages/Twisted-15.3.0-py2.7-linux-x86_64.egg/twisted/python/context.py", line 81, in callWithContext\n    return func(*args,**kw)\n', '  File "/data/seismo/seiscomp3-jakarta-2016.333/lib/python/seiscomp3/fdsnws/event.py", line 628, in _processRequest\n    return self._processRequestExp(req, ro, dbq, exp, ep)\n', '  File "/data/seismo/seiscomp3-jakarta-2016.333/lib/python/seiscomp3/fdsnws/event.py", line 334, in _processRequestExp\n    objCount += self._loadComment(dbq, e)\n']`

gempa-stephan, 2017-02-24:
Indeed the implementation includespicks per default. The FDSN Spec says nothing about picks but defines the default for includearrivals to false. I think it is better to change the default in the implementation from true to false. This needs to be discussed.

Regarding the includecomments, yes please file a bug report.

jordi-domingo, 2017-02-24:
Ok, then I'll do the pull request assuming includearrivals is false by default.

gempa-stephan, 2017-02-24:
Yes, go ahead. Meanwhile I fixed the includecomment but locally. I will push the solution soon.

closed 2017-02-27 by jordi-domingo

gempa-stephan, 2017-02-27:
The default for includepicks remains to be `true`. However, picks will only be available if arrivals are requested. 01cb9b0 corrects the default value in the WADL, HTML and URL builder and adds a description of the arrival dependency.

jordi-domingo, 2017-02-27:
Ok, fine!