SeisComP3/seiscomp3 #102: fdsnws - includecomments=true cause error

jordi-domingo, 2017-02-27:
An explicit `includecomments=true` seems to not work, see log below:

`2017/02/24 13:33:46 [error/log] FDSNEvent instance has no attribute '_loadComment' [' File "/home/sysop/.local/lib64/python2.7/site-packages/Twisted-15.3.0-py2.7-linux-x86_64.egg/twisted/python/threadpool.py", line 196, in _worker\n result = context.call(ctx, function, *args, **kwargs)\n', ' File "/home/sysop/.local/lib64/python2.7/site-packages/Twisted-15.3.0-py2.7-linux-x86_64.egg/twisted/python/context.py", line 118, in callWithContext\n return self.currentContext().callWithContext(ctx, func, *args, **kw)\n', ' File "/home/sysop/.local/lib64/python2.7/site-packages/Twisted-15.3.0-py2.7-linux-x86_64.egg/twisted/python/context.py", line 81, in callWithContext\n return func(*args,**kw)\n', ' File "/data/seismo/seiscomp3-jakarta-2016.333/lib/python/seiscomp3/fdsnws/event.py", line 628, in _processRequest\n return self._processRequestExp(req, ro, dbq, exp, ep)\n', ' File "/data/seismo/seiscomp3-jakarta-2016.333/lib/python/seiscomp3/fdsnws/event.py", line 334, in _processRequestExp\n objCount += self._loadComment(dbq, e)\n']`

gempa-stephan, 2017-02-27:
Fixed by 2a73058

jordi-domingo, 2017-02-27:
Great, thanks!

closed 2017-02-27 by jordi-domingo