SeisComP3/seiscomp3 #104: BUG/ENH: scautoloc with historical playback and disabled networks/stations

luca-s, 2017-03-10:
I am running scautoloc in playback mode on some historical data and I noticed that scautoloc fails to use stations that are not active now but that were active at that time.

I had a look at the code and the problem is in [here](https://github.com/SeisComP3/seiscomp3/blob/master/src/trunk/apps/processing/scautoloc/app.cpp#L391) and [here](https://github.com/SeisComP3/seiscomp3/blob/master/src/trunk/apps/processing/scautoloc/app.cpp#L401).

I am wondering if we could make the check on the network/station end date only if config.playback is disabled.

Thanks

luca-s, 2017-03-10:
Related to #23 but this is a proposal to make the check on end date optional (depending on playback flag)

gempa-jabe, 2017-03-10:
Actually just ignoring end time is not the right thing to do. If the station location was different at that time then it will not work. scautoloc has to lookup the correct network/station epoch for the time of the pick that it is processing. Another option is to pass a reference time along with the playback option but that is sometimes hard to figure out for a script, especially when you do a waveform playback. @jsaul

luca-s, 2017-03-10:
This makes sense and it 's probably too much effort to fix this particular case. I think a good improvements would be to log more information ("discarting this network/station because end date is ....") so that users can have a clue on what is happening inside scautoloc without looking at the code.

jsaul, 2017-03-10:
Thank you for reporting this. I agree that it would be best to implement 
this by adding a parameter --playback-reference-time or so in order to 
make it work correctly in the case of changing station coordinates (not 
uncommon in temporary deployments).

luca-s, 2017-03-10:
--playback-reference-time would be a really good option but I would keep the idea of logging the station discarding anyway.

gempa-jabe, 2018-11-01:
Fixed.

closed 2018-11-01 by gempa-jabe