SeisComP3/seiscomp3 #107: fdsnxml2inv-import_inv dlsv : old date rejected

lemarchandarnaud, 2017-03-31:
Dear Jan, 
Fdsnxml2inv import tool rejects two stations of the  StationXML file (20161202OVSGSources.xml) because their startingDate  are  too old. The 2 stations with old startdate are discarded. 
 [warning] L1666: (optional) Station.CreationDate: invalid value  '1847-01-01T00:50:39.000' in CreationDate: non optional member////22:27:58 
[warning] L6674: (optional) Station.CreationDate: invalid value '1797-01-01T00:50:39.000' in CreationDate: non optional member/

The import_inv dlsv tool doen't reject the station but set the start date to 1970-01-01 without any warning. 

This is an issue , we didn't discuss  last January, but indeed volcano observatories  have very old 
data I would like to archive. In this case this is a temperature of thermal springs monitored by the French colons 200 years ago !

gempa-jabe, 2017-03-31:
SeisComP3 uses epoch timestamps that are the seconds since 1970. On 64bit systems we can handle such timestamps as yours but on 32bit systems it won't work. Even worse, you can't catch errors in a stable manner.  The only stable time range is 1970-2037. So supporting that would require some fundamental changes for handling timestamps in SC3. For SEED I could think of a hack to support it but for FDSNXML I don't see a stable way.

lemarchandarnaud, 2017-03-31:
Thus this is a limitation  I must aware of. It concerned few stations and few data.
Thanks

closed 2017-05-02 by gempa-jabe