SeisComP3/seiscomp3 #110: fdsnxml2inv - import_inv dlsv : names of encoding format

lemarchandarnaud, 2017-03-31:
Dear Jan
When dataless files are converted into inventoryXML, the format of data is steim2 whatever the encoding format  in the dataless. Knowing that this information is not included in the stationXML anymore( it shoud not be in  meta-data, it was the conclusion of our meeting last January), I suggest not to fill the field format  in the inventory  even if the information is in the dataless SEED. 
For any reasons if you want to fill it with the dlsv tool, I see two possibilities.
Either  you  use the "short descriptive  name"  of the blockette 30 which describes the format,  or you try to map this   "short descriptive  name"  on  the codes used  in the blockette  1000.
See you
arnaud

gempa-jabe, 2017-05-02:
The fixed format has been removed and the short descriptive name is used instead if given.

lemarchandarnaud, 2017-05-19:
When I convert dataless file in inventory xml (branch dlsv-conversion-fix) ,  I still have this message "[warning] No mapping to a name from decoder keys known: M0; W4 D0-31 C2" for integer data format (not steim2). I don't see the short descriptive name that you mentioned in your fix. Where is it implemented, in the stream description?

gempa-jabe, 2017-05-22:
I have committed a patch. It should now populate the short name in the format attribute. Note that currently the SC3 database schema limits the length of the format string to 20 characters whereas SEED allows 50 characters. This needs to be fixed as well.

closed 2017-12-04 by gempa-jabe