SeisComP3/seiscomp3 #112: fdsnxml2inv - import_inv dlsv : empty gainUnit in streams  for SOH channel : UBE (battery voltage)

lemarchandarnaud, 2017-03-31:
Dear Jan
In my dataless files, I have  a state-of-health channel (UBE:battery voltage). When I convert it in the inventory, no sensor is associated to this channel.
In January you explained to me that stages are either affected to sensor or datalogger. Indeed the import tool parse stage input units of the stages to detect the first stage with volts (or amps) to get the first stage of the datalogger. For my state of health stream, the very first stage matched this criteria so no sensor is detected.
But when it happens, the stream has no “gainUnit” although I expected Volts in this field.

Here is your answer ... sorry I speak for you !
I was also thinking about such as case. Because now Volts or Ampere are used as eletric input/output (the connection between sensor and datalogger or inside datalogger) they are more or less excluded as sensor unit. If the sensor monitors voltage then we have a small problem here. I will check if that test can be made more stable. In FDSNXML the channel carries a sensor unit in instrumentSensitivity. Not sure right now if SEED does the same for stage 0. Otherwise one could try to figure that out based on analog and digital  filters and there order. Maybe you come up with a simple idea!?

Now, Arnaud is writing ....
I confirm that in the dataless SEED, the stage0 is an additional stage with only a gain defined for a normalized frequency, it carries no unit.The input unit of channels  is defined in the channel attributes and in stage 1. I entertain this is a duplicated information. Nevertheless the output unit must  be assessed from the output unit of the last stage.
In the StationXML , the InstrumentSensitivity element ( in responses elements) includes the input and output units but this is redundant with units described in all stages.
As I told you in January, I think it would be more natural to include all analog stages in the sensor and digital stage in the data-logger. I know this is questionable because dataloggers often have analog pre-amplier which would be in the sensor part. Sensors also has pre-amplifiers which are  currently in datalogger part.

gempa-jabe, 2017-05-04:
Can you provide an example SEED file with UBE channel?

lemarchandarnaud, 2017-05-24:
I sent you last one last Monday (5/22/2017):  20161202OVSGInc.dataless

closed 2017-12-04 by gempa-jabe