SeisComP3/seiscomp3 #114: inventory  enhancement:  unit description

lemarchandarnaud, 2017-03-31:
Dear Jan

All my unit codes are correctly handled, I didn't see any errors in my tests. Nevertheless, the description in daatless SEED and StationXML is lost in the inventory. The code is often clear enough ( MM for millimeters, KM/H, HPA for hectopascal). But I have others codes more difficult to guess : DB for decapascal, DEG for degree of an angle, C for degree Celsius, PDB for “Carbone 13 fraction - Pee Dee Belemnite”.
I think that a description of units is very useful. I  don't know if you can find a solution to the description of units.
Regards.

arnaud

gempa-jabe, 2017-04-03:
Actually the current database model only supports 20 characters per unit. I would strongly advise to use standard units. We could use the ```remark``` field in Sensor to describe special units that are not common and known by the conversion tool.

lemarchandarnaud, 2017-04-05:
Dear Jan 
I  agree with you, standard units are recommended but sometimes one  doesn't have the choice.  If you use the remark field in the Sensor, you must keep in mind that the description (Sensor remark) should also be exported in stationXML files  get by a fdsn web-service request.

gempa-jabe, 2017-04-05:
Sensor.remark != Sensor.description. Both attribute are available in SC3 and only Sensor.description is exported to FDSNXML. The remark attribute is SC3 only.

lemarchandarnaud, 2017-04-12:
So have you a solution to export the unit description to FDSNXML from  SC3? Indeed I think this is a critical information for users when the unit is not common.

gempa-jabe, 2017-05-02:
I have added code to dlsv2inv and fdsnxml2inv that sets a JSON document as the Sensor.remark which contains the unit description. That way it can be extended with other remarks.

closed 2017-12-04 by gempa-jabe