SeisComP3/seiscomp3 #115: fdsnxml2inv - import_inv dlsv :  description / model / manufacter

lemarchandarnaud, 2017-03-31:
Dear Jan
The import_inv dlsv fills  description, model and manufacter (attribute of datalogger)  according to   the information in the IntrumentDescription of the blockette 52 which is a cross reference to the dictionary of generic abbreviations (blockette 33). In the dictionary, the instrument is described with a short phrase.

I  expected that this short phrase would be in the description attribute of dataloggers . However the import_inv tool parses the phrase to fill description, model and manufacter fields in the inventory.
When I convert the dataless SEED in StationXML, the description ( child element of sensors)  in  stationXML  matches the description of the sensor field (blockette 33).

Thanks for your help
arnaud

gempa-jabe, 2017-04-03:
> The import_inv dlsv fills description, model and manufacter (attribute of datalogger) according to the information in the IntrumentDescription of the blockette 52 which is a cross reference to the dictionary of generic abbreviations (blockette 33).

If you refer to field 6 then this is an instrument identifier, not a description. And it refers to the sensor not to the datalogger in my understanding. The datalogger description is currently set to the dataloggers name which is redundant. I will remove that in branch ```dlsv-conversion-fix```.

gempa-jabe, 2017-04-03:
74071d27ee0ae5b1ad960cb7b8f2dca3a8883e68 implements that the full description is written into Sensor.description.

lemarchandarnaud, 2017-04-05:
Dear Jan
Yes you are right,  I made a confusion .  I meant sensor description instead of  datalogger description.
The filed 6 is a  id which points to an entry of the dictionary. This entry is a instrument description. 
I test yours modifications. The instrument description is fine but the model and the manufacturer are still filled with inconsistent strings.
thanks

gempa-jabe, 2017-04-05:
> The instrument description is fine but the model and the manufacturer are still filled with inconsistent strings.

There is nothing to fill with from SEED. The initial ```dlsv2inv``` implementation tokenizes the string and sets the model and manufacturer. That code is not mine and does it hurt to set those values?

gempa-jabe, 2017-04-05:
>  does it hurt to set those values?

Common cases with seismic instruments work quite well. But the SEED files for which I have a copy from you do not work. I agree that the values are nonsense but I have no idea how to fix that and to keep the old (rather nice) behaviour.

lemarchandarnaud, 2017-04-12:
I will adapt my dataless to the seismic instrument. I just need to know how the converter parse the string. I mean what are separators and where model description and  manufacturer informaton are supposed to be. I can look at the code to see that. Do you know if a documentation exist for this point?

gempa-jabe, 2017-04-12:
No, there is no documentation. The code starts [here](https://github.com/SeisComP3/seiscomp3/blob/dlsv-conversion-fix/src/trunk/apps/tools/inventory/dlsv2inv/inventory.cpp#L2340).

lemarchandarnaud, 2017-04-12:
ok thanks. The description can be :   manufacturer' 'model[/type]' ' blabla ...
I will  change all my descriptions to match seismic convention.

closed 2017-04-12 by lemarchandarnaud