SeisComP3/seiscomp3 #116: fdsnxml2inv - import_inv dlsv :  gains of data-loggers

lemarchandarnaud, 2017-03-31:
Dear  Jan
The gain attribute of dataloggers seems to be equal to the gain of the first stage detected as a part of the datalogger. I would expect that the gain is the multiplication of all the gain of stages included in the
datalogger.

I speak for Jan who already answered me :
The gain of the datalogger is used exclusively if no further stages are defined otherwise it will be multiplied with all other stages, a pre-amplifier gain. Usually SEED uses dummy responses to define this pre-amplifier gain. We try to detect that case and set the datalogger gain. Without pre-amplifier gain the datalogger gain is 1.

I am back ...:
I don't really understand your answer. In digging in the SC3ML of a complex seismic station I tried to clarify the meaning of the attribute "dataloggerGain". Either the import tool  detect a pre-amplifier with a "true" gain (not equal to 1 to reject dummy stages), in this case the dataloggerGain is the gain of the pre-amplifier , or no pre-amplifier is detected and then the dataloggerGain is the multiplication of all stages affected to the datalogger. Am I correct?
If yes, I would say that this is not very consistent but this not my business because this is not an information in the StationXML or datalessSEED.  But it is logical that 
sensorgain * dataloggergain = streamgain

Regards
arnaud

gempa-jabe, 2017-04-03:
> Either the import tool detect a pre-amplifier with a "true" gain (not equal to 1 to reject dummy stages), in this case the dataloggerGain is the gain of the pre-amplifier , or no pre-amplifier is [...]

No. 
```
Stream.gain = Sensor.response.gain * Datalogger.gain * Datalogger.stage1.gain 
* Datalogger.stage2.gain * ...
```
 So in any case you multiply the gains of all stages. In that evaluation there is no **if**. Either the Datalogger.gain is 1.0 or something else. @andres-h recommends to set the gain to 1.0 if there is a stage that has a gain != 1.0. If all stages have a gain of 1.0 then you can set the Datalogger gain to the overall gain of the digitizer. Anyhow, the final Stream gain is computed as described above. Setting the datalogger.gain to the preamplifier gain (if any) is done by the conversion tools.

lemarchandarnaud, 2017-04-05:
Dear Jan 
These  sentences "@andres-h recommends to set the gain to 1.0 if there is a stage that has a gain != 1.0. If all stages have a gain of 1.0 then you can set the Datalogger gain to the overall gain of the digitizer."   confused me. Thus I made tests with a dataless SEED with 4 stages:
Sensor (stage1) * pre-amp(stage2) * digitizer(stage3) * FIR (stage4). 

First of all, if the stage0 is omitted in the dataless, the stream.gain is left empty in the inventory. If  a stage0  is inserted with a dummy gain (3.14) the stream.gain = 3.14 in the inventory . Thus the  dlsv doesn't  evaluate the overall gain of channel :   Stream.gain = gain of stage 0 if it exits.

case1:  Sensor (stage1) * pre-amp(stage2) * digitizer(stage3) * FIR (stage4)=  120 *  1 * 1e6 * 4
Sensor.gain= 120 (ok)
Datalogger.gain= 1e6 ( not set to 4*1e6 ? )

case2:  Sensor (stage1) * pre-amp(stage2) * digitizer(stage3) * FIR (stage4)=  120 *  0.5 * 1e6 * 4
Sensor.gain= 120 (ok)
Datalogger.gain= 0.5

gempa-jabe, 2017-04-05:
> First of all, if the stage0 is omitted in the dataless, the stream.gain is left empty in the inventory. If a stage0 is inserted with a dummy gain (3.14) the stream.gain = 3.14 in the inventory

Isn't that what a conversion tool is supposed to do: to convert? And that is what it does. Transform information from one format into another without trying to change information.

> case1: Sensor (stage1) * pre-amp(stage2) * digitizer(stage3) * FIR (stage4)= 120 * 1 * 1e6 * 4
Sensor.gain= 120 (ok)
Datalogger.gain= 1e6 ( not set to 4*1e6 ? )

The gain of 4 is part of the decimation and not the datalogger object itself. If you would add another decimation to the same datalogger that does not have a FIR stage with gain 4 then it would not work. Here SEED is completely different from SC3. One datalogger can hold a setup for many output sample rates.

> case2: Sensor (stage1) * pre-amp(stage2) * digitizer(stage3) * FIR (stage4)= 120 * 0.5 * 1e6 * 4
Sensor.gain= 120 (ok)
Datalogger.gain= 0.5

Same here. The preamplifier stage is (hopefuly) common for all decimation stages and therefore set as datalogger gain. The idea is to describe the instrument not just one particular channel setup. The 0.5 could also go into the decimation analogue or digital filter chain as a dummy response and the datalogger gain would be set to 1.0.

lemarchandarnaud, 2017-04-12:
"Isn't that what a conversion tool is supposed to do: to convert?"
Yes I agree, but if for any reason the stream gain is missing of fake  in the dataless or StationXML,  converters could (re)evaluated it ( with a warning of course). But no try to change or enhance information doesn't shock me.

"Here SEED is completely different from SC3. One datalogger can hold a setup for many output sample rates ... The idea is to describe the instrument not just one particular channel setup"
I don't really see that in the array of dataloggers in SC3. Indeed each setup seems to create a new datalogger in the list of dataloggers. In others words  this datalogger is specific to particular channel setup. It can be reuse for different channels with the same setup.

I understand that the datalogger.gain is either a pre-amp gain (dummy stage excluded) or the digitizer (Count/volt) itself. If this datalogger.gain is a pre-amp gain a FIR (with a decimation.factor=1) in the DigitalFilterChain  describes the digitizer gain. I send you a file with 3 channels (email). The two first ones are quite similar except the pre-amp. The first one has a pre-amp=1 and the second the pre-amp=0.25.
When I convert it with dlsv2inv ,both channels have the same datalogger description with a digitizer.gain=0.25 . I am wandering if this is a bug.

I think that the pre-amp shoulb be in the AnalogFilterChain as a PAZ without zeros and poles. Thus the datalogger.gain would always be the digitizer gain (Count/V) and nothing else.
It would be less confusing and more clear.

gempa-jabe, 2017-05-02:
See 70814988255341030e40f7ce195229a97ee4021a.

lemarchandarnaud, 2017-05-23:
here is how I  see the definitions:
Pre-amp:     analog stages  with no pole or zeros. It occurred  before the
digitizer stage ( I define it after). Pre-filter:  analog stages  with
poles and zeros.  The gain of this stage is often 1. they occur  before the
digitizer stage. In some cases, pre-filter stage has a gain!=1. Thus it can
be seen as a mixed of pre-amp and pre-filter.

The digitizer stage  is  the analog to digital converter (ADC) itself  which
defines the conversion of V into count. So I suggest  the field gain in the
datalogger element in the inventory, should be equal to the digitizer (ADC)
gain.

closed 2018-11-01 by gempa-jabe