SeisComP3/seiscomp3 #118: fdsnxml2inv:  Polynomial responses rejected

lemarchandarnaud, 2017-04-12:
Dear Jan
dlsv2inv deals with polynomial responses very well  (dlsv-conversion-fix branch) , but fdsnxml2inv does not. No  polynomial  are  inserted at all.  Streams.datalogger  and streams.sensor  are  left empty.

Here is interresting  lines of the log of the converter:
 01:04:46 [warning] L508: Stage: missing mandatory attribute: StageGain
01:04:46 [info] L508: Invalid Stage element: ignoring
01:04:46 [warning] L878: Stage: missing mandatory attribute: StageGain
01:04:46 [info] L878: Invalid Stage element: ignoring

Polynomial responses seems not recognized.
Thanks for your help

gempa-jabe, 2017-04-13:
The StageGain is mandatory in the ResponseStageType of FDSN StationXML, see http://www.fdsn.org/xml/station/fdsn-station-1.0.xsd. Nothing I can do about it. The parser ignores those invalid structures. You would have to use a StageGain of 1.

lemarchandarnaud, 2017-04-13:
First of all, the StationXML file , I am testing, is an output of stationxml-converter-1.0.9 (iris). The input is a dataless SEED that I use to test dlsv2inv.  Both StationXML and dataless have the same information.

The problem occurs when the stage1 is  a polynomial response (Line 508 and 878 are the beginning of stage 1). In this case the stage can' t have a StageGain (equivalent of Blockette 58 in SEED). I never add a blockette 58 (gain) to a stage of a polynomial response because it does not make sense.
If I try, verseed ouputs a warning and stationxml-converter-1.0.9 inserts  2 stage 1 in the StationXML !
But indeed if I edit the statioxML file to add a dummy StageGain to the polynomial Response  in the stage1, it works. 
The documentation of "ResponseStageType" states: "This complex type represents channel response and covers SEED blockettes 53 to 56". 
So I am wandering if there is a bug because  the polynomial response is decribed by the  blockette 62.  The option minOccurs="0" might be added to the StageGain element (like the decimation element) ?

gempa-jabe, 2017-05-02:
There are possibly bugs in the schema documentation and definition but that cannot be handled in this issue. I have implemented the parser against the provided schema. If the schema is not applicable for some use-cases then it should be fixed. You should talk to IRIS or FDSN in that case. We are not the maintainers of StationXML.

lemarchandarnaud, 2017-05-19:
I already posted an issue on the github of the FDSN: [https://github.com/FDSN/StationXML/issues/13#issuecomment-297932494].
This is a bug already known and it should be fixed in the next version of the schema.
Is there a possibility to force the parser  to use an alternative schema?

gempa-jabe, 2017-05-22:
> Is there a possibility to force the parser to use an alternative schema?

No, the schema parser is in the source code and not an external xsd. We would have to fix the code.

gempa-jabe, 2017-05-22:
I committed a less strict parser version. You can check it out. It should handle stageGain as optional.

closed 2018-11-01 by gempa-jabe