SeisComP3/seiscomp3 #132: Case insensitive header check

junghao, 2017-07-13:
We've met some services that returns http headers in lower case. 
The lower case headers cause this header comparison fails :
https://github.com/SeisComP3/seiscomp3/blob/master/src/trunk/libs/seiscomp3/io/recordstream/fdsnws.cpp#L234

Since HTTP spec defines headers to be case insensitive so a case insensitive comparison should be preferred.

gempa-jabe, 2017-07-13:
Your are right and we have to fix that. Thank you.

junghao, 2017-07-13:
Thanks. I think there're some other header checks have to be fixed as well.

gempa-jabe, 2017-07-13:
Correct. Furthermore https support and redirects are missing. That will be implemented as well while fixing the code.

gempa-jabe, 2017-07-13:
I committed a fix. Please give it a try. It supports https through ```fdsnwss://``` (note the trailing 's' for SSL) as well.

junghao, 2017-07-13:
Yes thanks it works.

However my compiler doesn't like this `typename`
src/gui-qt4/apps/scmv/eventdata.cpp:218: error: using ‘typename’ outside of template

After I removed it then it compiles.

gempa-jabe, 2017-07-14:
Thank you for confirmation. I close the issue. I have also fixed the scmv compilation issue.

closed 2017-07-14 by gempa-jabe