SeisComP3/seiscomp3 #133: scmv removeEventDataOlderThan seems not to be working

a-oth, 2017-07-13:
Hi,
In scmv, it seems that event data are not removed anymore from the map after removeEventDataOlderThan seconds (we are running the latest commit from 7 July). Can you confirm if this is a bug or if it is rather somehow related to our setup?
Thanks very much!

gempa-jabe, 2017-07-13:
Hi, it works for me. How did you verify the bug? It is hard for me to reproduce and follow a "it seems that ..." report ...

a-oth, 2017-07-13:
Hi, thanks for the quick answer. For instance, I have set removeEventDataOlderThan to 10800. After a new event has arrived, it is not removed from the map again 3 hours later. This happens independently of whether the event was read from the database when starting scmv or if it was received my the messaging system. If we let scmv run, we simply accumulate all new events without that older ones get removed following the removeEventDataOlderThan setting.

gempa-jabe, 2017-07-13:
Does that also hold for the event table?

a-oth, 2017-07-13:
I hadn't looked at the event table so far in detail, but for the latest event which should already have been removed from the map, it is also still present in the event table.

gempa-jabe, 2017-07-13:
OK, thanks, I will check the behaviour again.

gempa-jabe, 2017-07-13:
Note that events are only removed when a event is added. If no event is added for 1 day then even your 3 hours event buffer is not updated and therefore no event is removed from the list / map.

a-oth, 2017-07-13:
Ok, thanks, I did not realise that. I just gave a look again and it looks alright now with my 3-hours example. We actually just had an event that arrived in this very moment roughly 8 hours after the last one and the latter has now indeed been removed.

gempa-jabe, 2017-07-13:
If that behaviour is confusing then we can add code to check for expired events every 10 seconds or so. For me it hasn't been an issue so far.

a-oth, 2017-07-13:
I think it just confused me now since I noticed by coincidence that something was not working following what I expected from the removeEventDataOlderThan parameter, but it's not really an issue in that sense, so from my side, it can definitely stay as it is. Many thanks!

closed 2017-07-13 by gempa-jabe