SeisComP3/seiscomp3 #134: Changing Arclink user for recordstream

Jollyfant, 2017-07-27:
Hi, we are running scwfparam that requests data from recordstream Arclink. In the logs these requests show up as guest@anywhere. Can we configure this username anywhere?

Best,
Mathijs

andres-h, 2017-07-27:
See https://docs.gempa.de/seiscomp3/current/apps/global_recordstream.html#arclink

Jollyfant, 2017-07-27:
Hi Andres, so putting this in global.cfg would work?

    recordstream.source = eida.orfeus-eu.org:18001?user=rrsm

andres-h, 2017-07-27:
Yes, it would work in global, but then all programs use this username. 
Might be better to put it in scwfparam.cfg.

Maybe also include the domain (useful if the logs are combined), eg.:

recordstream.source = eida.orfeus-eu.org:18001?user=rrsm@orfeus-eu.org

Jollyfant, 2017-07-27:
Ok cheers

closed 2017-07-27 by Jollyfant