SeisComP3/seiscomp3 #135: Arclink server issue using scolv from a client

a-oth, 2017-07-27:
Hello,
We noticed that there is an issue with the Arclink server when using arclink in the recordstream for scolv. I attach here an example request file and the arclink log file. The request has size="0" for all records and other parameters "UNSET".

The log states that "RuntimeError: Option not found for: reqhandler.trackdir". If I set this option, then it gives the same message for option reqhandler.filedb respectivel reqhandler.subnodelist. Even if I set values for all of these, I still get an "unexpected eof cmd" error.

The problem does not appear in the 2017.124.02 release, there all works fine. From the master branch, we tested commits 3f94e4f (the latest one) and 15689f8, both with the issue occurring.

Did anybody else notice this issue thus far?
Many thanks!
[arclink.txt](https://github.com/SeisComP3/seiscomp3/files/1180196/arclink.txt)
[request_example.txt](https://github.com/SeisComP3/seiscomp3/files/1180198/request_example.txt)

gempa-jabe, 2017-07-27:
Thanks a lot for bringing up this issue. This is due to the Python wrapper changes which are very intrusive. We will check and fix it. I will report back when you can test again. Thanks again.

a-oth, 2017-07-27:
Many thanks, ok!

a-oth, 2017-07-28:
OK, thanks.

a-oth, 2017-07-28:
Hi, we justed tested the fix, it works fine. Many thanks for resolving this issue!

gilcel, 2017-07-28:
To be be precise it was fixed with commit 473a168f76f5612a96912f6b7b229f12c4c7c0e2

closed 2017-07-31 by gempa-jabe