SeisComP3/seiscomp3 #136: scwfparam values

Jollyfant, 2017-08-01:
Hi devs, we were updating an old installation of SeisComP3 (2015.149) to the current stable release. Running scwfparam with the same configuration gives consistently **lower** peak motion values compared to the old installation (in the order of ~1%).

I'm wondering, has there been any changes or bug fixes that may be behind these differences?

gempa-jabe, 2017-08-01:
Actually there were some fixes but I don't know of any that should have an effect on the peak amplitudes. What has been done in general is to replace the Butterworth bandpass with a chain of highpass and lowpass filters by default. But only if you configure a filter by string, e.g. "BW(...)" which now is an alias for "BW_HLP(...)" rather than "BW_BP(...)". But that should not affect scwfparam since it creates the filters explicitely in the code. You can check https://github.com/SeisComP3/seiscomp3/blame/master/src/sed/apps/scwfparam/processors/pgav.cpp for changes. Do you apply filters causal or acausal?

Jollyfant, 2017-08-01:
Filters are acausal bandpass but I'm pretty confident the filters are OK.

I noticed from the logs that occasionally there seems to be a discrepancy in the gain value. It appears that the gain in the old case is 624830000 as can be seen in the station inventory from NOA (http://eida.gein.noa.gr/fdsnws/station/1/query?station=KOSK&level=response&channel=HHN). However, if you calculate the total gain through the stages it adds up to 625000000. So maybe there was a change in how the total gain is determined and used by scwfparam?

Old:

    2017/07/31 06:44:29 [debug/PGAV] > processing HC.KOSK..HHN
    2017/07/31 06:44:29 [debug/PGAV] >  cut 2.08s from beginning
    2017/07/31 06:44:29 [debug/PGAV] >  trigger = 2017-07-30T07:02:24.464707Z
    2017/07/31 06:44:29 [debug/PGAV] >  time window = 2017-07-30T07:02:04.464999Z ~ 2017-07-30T07:03:29.464999Z
    2017/07/31 06:44:29 [debug/PGAV] >  max counts = 363698.0
    2017/07/31 06:44:29 [debug/PGAV] >  gain = 624830000.00

New:

    2017/07/31 06:43:40 [debug/PGAV] > processing HC.KOSK..HHN
    2017/07/31 06:43:40 [debug/PGAV] >  cut 2.08s from beginning
    2017/07/31 06:43:40 [debug/PGAV] >  trigger = 2017-07-30T07:02:24.464707Z
    2017/07/31 06:43:40 [debug/PGAV] >  time window = 2017-07-30T07:02:04.464999Z ~ 2017-07-30T07:03:29.464999Z
    2017/07/31 06:43:40 [debug/PGAV] >  saturation threshold = 6710886.400000
    2017/07/31 06:43:40 [debug/PGAV] >  max counts = 363698.0
    2017/07/31 06:43:40 [debug/PGAV] >  gain = 625000084.77

Best,
Mathijs

gempa-jabe, 2017-08-01:
See here: https://github.com/SeisComP3/seiscomp3/blame/master/src/trunk/libs/seiscomp3/processing/stream.cpp#L173

There is now a new gain correction if the sensor gain frequency differs from the overall gain frequency. In prior versions we corrected the gain for 1 Hz with respect to the given gain frequency, but that was very wrong is some cases. This check was disabled in 2015.149. Please check for debug logs like "correct gain by factor ..." indicating the correction.

Jollyfant, 2017-08-01:
Doesn't look like there is an entry in the logs. It should be in the scwfparam logs I suppose? Logging level is set to DEBUG.

gempa-jabe, 2017-08-01:
Yes, scwfparam.log. It should appear before the actual processing log output. Feel free to add more debug output to ```Seiscomp::Processing::Stream::init```. It assigns the inventory gain (which you should check in your inventory for comparison) and then applies the correction if necessary.

Jollyfant, 2017-08-01:
Sorry I checked again and updated my comment..

gempa-jabe, 2017-08-01:
OK, so you have the reason for the difference. And now we can enter the discussion about what is correct and what not ;)

gempa-jabe, 2017-08-01:
Can we close the issue?

Jollyfant, 2017-08-01:
Yeah, I'll close but unfortunately I opened a new one with some more questions :-)

closed 2017-08-01 by Jollyfant