SeisComP3/seiscomp3 #137: scwfparam questions

Jollyfant, 2017-08-01:
Hi devs, I had some more questions about scwfparam. Sorry it is a really long list but I am a bit stumped by the entire thing.

* When an event is dispatched through scdispatch it is processed by scwfparam. What happens when we dispatch the same event with `-O update` because the origin has changed? From the logs I do not see that all channels are being reprocessed, but only the "remaining" ones. And old streams are stuck using incorrect filters (as they are magnitude dependent). On a related note, is the re-processing scheduling reset after an update is sent (as configured `wfparam.cron.delayTimes`).

* If we pass `-O remove` to scdispatch with an event are the strong motion values automatically removed? Is the journal entry removed?

* Is there a way to force scwfparam to **always** reprocess the entire thing when an update comes through? I saw the `--force` flag but I am unsure that does what I want. I don't think caching anything is required since the amount of data is negligible and the journal entry prevents events from being processed again:

```
2017/08/01 06:49:29 [info/WfParam] 20170731_0000066: found journal entry "completely processed", ignoring event
```

* The output processed waveforms are sometimes scattered over multiple directories for the same event (because the origin has changed; `3.8` to `4.0`). In essence the same question as my first but some streams are present in one and not the newest folder. Is it because they are not reprocessed, or should I take the files from the "old" origin folder?

    20170731051902_38_34320_025640_20170731063912
    20170731051905_40_34380_025530_20170731053909

* Lastly, I checked my `scmaster.log` and it throws this exception quite consistently since we use `scdispatch` to update events. Looks like a field is `NULL` in the code that generated the query string or something. 
 
```
2017/08/01 10:53:04 [error/MYSQL] execute("update OriginReference set  where _parent_oid=20887851 and originID='Origin#SP20170801_0000042'") = 1064 (You have an error in your SQL syntax; check the manual that corresponds to your MariaDB server version for the right syntax to use near 'where _parent_oid=20887851 and originID='Origin#SP20170801_0000042'' at line 1)
```

gempa-jabe, 2017-08-01:
Actually this are usage questions and better placed at the forum at forum.seiscomp3.org. The Github issue tracker is meant to be used for code/development issues. At the SC3 forums are also the SED people, who requested and designed that module.

gempa-jabe, 2017-08-01:
Your last item seems to be a bug that I will investigate but has nothing to do with scwfparam itself.

closed 2017-08-01 by gempa-jabe

Jollyfant, 2017-08-01:
Ok thanks, Jan. I will post this on the forum.