SeisComP3/seiscomp3 #142: Picker doesn't show all stations with PostgreSQL database

Brtle, 2017-10-27:
Hi,

I have some difficulties in displaying all my stations in the picker. Some stations don't have picks yet and aren't visible when I click the `ShowAllStations` button. The signal response unit is M/S**2 so it doesn't pass [this condition](https://github.com/SeisComP3/seiscomp3/blob/master/src/gui-qt4/libs/seiscomp3/gui/datamodel/pickerview.cpp#L3984). Is there a reason to check only the `MeterPerSecond` unit and not the [other units](https://github.com/SeisComP3/seiscomp3/blob/master/src/trunk/libs/seiscomp3/processing/waveformprocessor.h#L64)?

By the way, I saw that the configured stations should be visible in the picker. I added a small binding to my station but [this condition](https://github.com/SeisComP3/seiscomp3/blob/master/src/gui-qt4/libs/seiscomp3/gui/datamodel/pickerview.cpp#L1255) is always false. I checked the values and it try to compare `HNZ` to `\x484e5a`, which is the hexadecimal representation of `HNZ`. Indeed, in PostgreSQL, the table `parameter` looks like:

```
select * from parameter;
  _oid  | _parent_oid |       _last_modified       |   m_name    | m_value
--------+-------------+----------------------------+-------------+----------
 294177 |      294176 | 2017-10-26 16:58:34.194214 | detecStream | \x46485a
 294178 |      294176 | 2017-10-26 16:58:34.200715 | detecLocid  | \x3030
 294160 |      294159 | 2017-10-26 16:58:34.2071   | detecStream | \x484e5a
 294165 |      294163 | 2017-10-26 16:58:34.211143 | detecLocid  | \x3030
 294164 |      294163 | 2017-10-26 16:58:34.215138 | detecStream | \x46485a
 294169 |      294168 | 2017-10-26 16:58:34.219155 | detecStream | \x484e5a
 294173 |      294172 | 2017-10-26 16:58:34.223252 | detecStream | \x484e5a
(7 rows)
```

I also tried with an SQLite database and everything is working fine. I can see my stations in the picker. The table looks fine:

```
select * from parameter;
581|580|2017-10-26 13:38:10|detecStream|HNZ
585|584|2017-10-26 13:38:10|detecStream|FHZ
586|584|2017-10-26 13:38:10|detecLocid|00
590|589|2017-10-26 13:38:11|detecStream|HNZ
594|593|2017-10-26 13:38:11|detecStream|HNZ
782|781|2017-10-26 13:55:54|detecStream|FHZ
783|781|2017-10-26 13:55:54|detecLocid|00
```

So I think there is an issue with the `m_value` field type of the `parameter` table in PostgreSQL.

gempa-jabe, 2017-10-27:
This is an issue with PostgreSQL >= 9. They changed the default output for bytea columns to hex. You have to enable the old representation with
```
ALTER DATABASE seiscomp3 SET bytea_output TO 'escape'
```

Brtle, 2017-10-27:
Damn, you're right! Besides, I already have some similar issues in the past… Thanks a bunch!

closed 2017-10-27 by gempa-jabe