SeisComP3/seiscomp3 #148: Hypo71 configuration

gempa-dirk, 2017-11-29:
@jmsaurel 

1. Is there any use of the Hypo71 plug-in parameter `defaultControlFile`?

2. If not, could the parameter be removed from the configuration? Additionally, could the first Hypo71 profile in the pull-down menu of scolv be the first profile in the `hypo71.profiles` parameter?

jmsaurel, 2017-11-29:
1. If not present, the plugins throws out a SEISCOMP_ERROR.
It is used to fill up variables with default values if they are not present in the profile.
For example, you could have a profile with only the mandatory values (velocity model and related stuff) and no TEST values definitions

jmsaurel, 2017-11-29:
2. Removed from configuration and hard-wired ?
Yes it could be, I don't remember why we choose to have it as a parameter.

gempa-jabe, 2017-11-29:
I think that configuration does not hurt and gives you more flexibility. There was a misunderstanding of the purpose of the parameter on our side. It is clear now, thanks.

closed 2017-11-29 by gempa-jabe