SeisComP3/seiscomp3 #149: Hypo71 default configuration

gempa-jabe, 2017-11-29:
Due to changes in parameter names (DISABLE_LAST_LOC -> !USE_TRIAL_POSITION), the default profiles should be adjusted as well:

```
profile.tectonic.conf:DISABLE_LAST_LOC = true
profile.volcanic.conf:DISABLE_LAST_LOC = false
```

Furthermore we should consider to default USE_TRIAL_POSITION to false if not given at all. See https://github.com/SeisComP3/seiscomp3/blob/master/src/ipgp/plugins/locator/hypo71/hypo71.cpp#L991.

CC @jmsaurel

gempa-jabe, 2017-11-29:
I created a PR to correct existing defaults. Would that be OK with you?

jmsaurel, 2017-11-29:
> Due to changes in parameter names (DISABLE_LAST_LOC -> !USE_TRIAL_POSITION), the default profiles should be adjusted as well:

profile.tectonic.conf:DISABLE_LAST_LOC = true
profile.volcanic.conf:DISABLE_LAST_LOC = false
 Ok for those

> Furthermore we should consider to default USE_TRIAL_POSITION to false if not given at all

The proposed modification is incorrect.
Either keep the condition as it is, 
or set `cCC.[lat1,lat2,lon1,lon2]` to `""` if `USE_TRIAL_POSITION` is `false`

gempa-jabe, 2017-11-29:
The condition is still correct, just the default value changes. It does not mean that the function always returns false rather than true. If you configure ```USE_TRIAL_POSITION = true``` then the function will return true. If you don't configure it at all, then the function will return false rather than true.

jmsaurel, 2017-11-30:
Agreed.

closed 2017-12-04 by gempa-jabe