SeisComP3/seiscomp3 #151: Arrivals not loaded in scamp

Brtle, 2017-11-29:
I tried to reprocess the amplitudes of an origin by sending the origin to `scamp` through the messaging system. According to the [source code](https://github.com/SeisComP3/seiscomp3/blob/master/src/trunk/apps/processing/scamp/amptool.cpp#L596), the origin is read by the `updateObject` method. However, the arrivals aren't loaded. The `scamp-processing-info.log` file looks like:

```
Processing report for Origin: Origin#20160416195532.829773.701501
-----------------------------------------------------------------
 + Arrivals
 + Nothing to do, no processors to add
 ```

 In the same time, with the `--origin-id` parameter, the [arrivals are explicitly loaded](https://github.com/SeisComP3/seiscomp3/blob/master/src/trunk/apps/processing/scamp/amptool.cpp#L277) before the origin processing. I tried to load the arrivals before [this line](https://github.com/SeisComP3/seiscomp3/blob/master/src/trunk/apps/processing/scamp/amptool.cpp#L599) and it's working fine with the messaging system:

 ```cpp
    Origin *origin = Origin::Cast(object);
    if ( origin ) {
        logObject(_inputOrgs, Time::GMT());
        query()->loadArrivals(origin);
        process(origin);
        return;
}
 ```

I don't fully understand the `scamp` module so was it intentional or a bug?

By the way, I didn't know where to put this this kind of report. Do you prefer that we open new topic on your forum or a new issue on Github?

gempa-jabe, 2017-11-29:
Since that is code related, a Github issue is just fine.

gempa-jabe, 2017-11-29:
How do you send the origin to scamp? It is not a bug since scamp has been working for years now.

Brtle, 2017-11-29:
I send the origin with a small Python custom module which looks like that:

```python
origin = self.db_query.loadObject(DataModel.Origin.TypeInfo(), origin_id)
origin = DataModel.Origin.Cast(origin)

DataModel.Notifier.Enable()
ep = DataModel.EventParameters()
DataModel.Notifier.Create(ep, DataModel.OP_UPDATE, origin)
DataModel.Notifier.Disable()

message = DataModel.Notifier.GetMessage(True)
self.connection().send(message)
```

gempa-jabe, 2017-11-29:
You just create an update notifier for the origin object. Now I fully understand your issue. scamp assumes that a new origin that is being received also comes with its arrivals. It does not react on late updates since we do not update origins with respect to their location. The use-case that you have in mind is currently not supported. I will push a fix (similar to yours) which should fix it. Can you give it a try?

Brtle, 2017-11-29:
Thanks a bunch! It's working great with your fix!

Is this fix will be available in the Jakarta 2017.320 release?

gempa-jabe, 2017-11-29:
Although the release will get another version number, yes. The next official release will have that fix.

closed 2017-11-29 by gempa-jabe