SeisComP3/seiscomp3 #152: Cache issue in scamp

Brtle, 2017-12-01:
Following [this issue](https://github.com/SeisComP3/seiscomp3/issues/151), I was able to generate new amplitudes with `scamp` by sending an origin using the messaging system. However, the old amplitudes wasn't recalculated due to [this condition](https://github.com/SeisComP3/seiscomp3/blob/master/src/trunk/apps/processing/scamp/amptool.cpp#L712). I so wanted to delete the old amplitude to let `scamp` regenerate them. I'm using a custom Python script which looks like that:

```python
ep = DataModel.EventParameters()

origin = self.db_query.loadObject(DataModel.Origin.TypeInfo(), origin_id)
origin = DataModel.Origin.Cast(origin)

DataModel.Notifier.Enable()

for i in xrange(origin.arrivalCount()):
    pick_id = origin.arrival(i).pickID()
    amplitudes = self.db_query.getAmplitudesForPick(pick_id)
    for amplitude in amplitudes:
        amplitude = DataModel.Amplitude.Cast(amplitude)
        # Quite ugly but otherwise the amplitude doesn't have parent
        DataModel.Notifier.Disable()
        ep.add(amplitude)
        DataModel.Notifier.Enable()

        ep.remove(amplitude)

DataModel.Notifier.Create(ep, DataModel.OP_UPDATE, origin)

DataModel.Notifier.Disable()

message = DataModel.Notifier.GetMessage(True)
self.connection().send(message)
```

At this point, the amplitudes are removed but `scamp` still find them:

```
2017/12/01 17:12:11 [info/AmpTool] Skipping MLv calculation for pick Pick#20170717074108.694751.41883, amplitude exists already
2017/12/01 17:12:11 [info/AmpTool] Skipping Md calculation for pick Pick#20170717074108.694751.41883, amplitude exists already
```

New amplitudes are generated if I restart `scamp` and rerun my script. It looks like a cache issue. Is it possible to flush this cache so that the amplitudes can be recalculated without restarting `scamp`?

gempa-jabe, 2017-12-04:
OK, another uncommon use case ;)
Actually scamp would need to remove an amplitude from its cache when it is removed. This hasn't been implemented because we usually do not remove objects. I will try to add it, stay tuned.

gempa-jabe, 2017-12-04:
I created the branch **scamp-remove-amp**. Can you test that?

Brtle, 2017-12-04:
I tried your branch and it worked as expected. Thank you!

gempa-jabe, 2017-12-04:
Perfect, then I will merge the changes Thanks for testing.

closed 2017-12-04 by gempa-jabe