SeisComP3/seiscomp3 #156: Missing TimeArray definition in SC3ML 0.10

Brtle, 2017-12-07:
The new SC3ML 0.10 introduces the [`TimeArray` complex type](https://github.com/SeisComP3/seiscomp3/blob/master/src/trunk/libs/xml/0.10/sc3ml_0.10.xsd#L231) based on the `qml:TimeArray` type. However, this type is not describe in the [quakeml_types.xsd](https://github.com/SeisComP3/seiscomp3/blob/master/src/trunk/libs/xml/quakeml_types.xsd) file.

This leads to an error when trying to read the XSD file with lxml:
```
>>> from lxml import etree
>>> etree.XMLSchema(etree.parse('sc3ml_0.10.xsd'))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "src/lxml/xmlschema.pxi", line 86, in lxml.etree.XMLSchema.__init__ (src/lxml/etree.c:204036)
lxml.etree.XMLSchemaParseError: complex type 'TimeArray', attribute 'base': The QName value '{http://quakeml.org/xmlns/quakeml/1.0}TimeArray' does not resolve to a(n) simple type definition., line 231
```

Can you add the new type to the `quakeml_types.xsd` file?

gempa-jabe, 2017-12-07:
It is not a new QuakeML type, just for SC3. So it must not be added to quakeml_types but the reference to QuakeML removed from the XSD. Does that patch work for you?

```
--- a/src/trunk/libs/xml/0.10/sc3ml_0.10.xsd
+++ b/src/trunk/libs/xml/0.10/sc3ml_0.10.xsd
@@ -230,7 +230,7 @@
   </xs:complexType>
   <xs:complexType name="TimeArray">
     <xs:simpleContent>
-      <xs:extension base="qml:TimeArray"/>
+      <xs:extension base=""/>
     </xs:simpleContent>
   </xs:complexType>
   <xs:complexType name="RealPDF1D">
```

Brtle, 2017-12-07:
Your patch does not work:

```
>>> from lxml import etree
>>> etree.XMLSchema(etree.parse('/home/clement/Code/seiscomp3/src/trunk/libs/xml/0.10/sc3ml_0.10.xsd'))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "src/lxml/xmlschema.pxi", line 87, in lxml.etree.XMLSchema.__init__ (src/lxml/lxml.etree.c:196366)
lxml.etree.XMLSchemaParseError: Element '{http://www.w3.org/2001/XMLSchema}extension', attribute 'base': '' is not a valid value of the atomic type 'xs:QName'., line 233
```
But `xs:string` can be used:

```diff
diff --git a/src/trunk/libs/xml/0.10/sc3ml_0.10.xsd b/src/trunk/libs/xml/0.10/sc3ml_0.10.xsd
index cfce34c..e6a30d3 100644
--- a/src/trunk/libs/xml/0.10/sc3ml_0.10.xsd
+++ b/src/trunk/libs/xml/0.10/sc3ml_0.10.xsd
@@ -230,7 +230,7 @@
   </xs:complexType>
   <xs:complexType name="TimeArray">
     <xs:simpleContent>
-      <xs:extension base="qml:TimeArray"/>
+      <xs:extension base="xs:string"/>
     </xs:simpleContent>
   </xs:complexType>
   <xs:complexType name="RealPDF1D">
```

The field will be very free but at least lxml won't complain.

gempa-jabe, 2017-12-07:
OK, we will need to generate a proper description of TimeArray.

gempa-jabe, 2017-12-07:
Here my proposal of quakeml_types with TimeArray:

```
<?xml version="1.0"?>
<xs:schema xmlns:xs="http://www.w3.org/2001/XMLSchema" xmlns="http://quakeml.org/xmlns/quakeml/1.0" targetNamespace="http://quakeml.org/xmlns/quakeml/1.0" elementFormDefault="qualified" attributeFormDefault="unqualified">
    <xs:simpleType name="ResourceIdentifier">
        <xs:restriction base="xs:string"/>
    </xs:simpleType>
    <xs:simpleType name="FloatArray">
        <xs:list itemType="xs:double"/>
    </xs:simpleType>
    <xs:simpleType name="ComplexArray">
        <xs:restriction base="xs:string">
            <!-- e.g. "(-0.01836,4.1e-05) (-0.01836,-4e-05) (-39.18,49.12) (-39.18,-49.12)"-->
            <xs:pattern value="(\s*\(\s*[+\-]?[0-9]+(\.[0-9]+)?([Ee][+\-][0-9]+)?\s*,\s*[+\-]?[0-9]+(\.[0-9]+)?([Ee][+\-][0-9]+)?\s*\)\s*)*"/>
        </xs:restriction>
    </xs:simpleType>
    <xs:simpleType name="TimeArray">
        <xs:list itemType="xs:dateTime"/>
    </xs:simpleType>
</xs:schema>
```

Brtle, 2017-12-07:
I just realized that FloatArray and ComplexArray wasn't in the QuakeML format. Why don't you put these types in your `sc3ml_0.10.xsd`? I don't see the point to having the `quakeml_types.xsd` file when you can have only 1 file.

By the way, your modification works well! I can validate SC3ML files with it.

gempa-jabe, 2017-12-07:
> I just realized that FloatArray and ComplexArray wasn't in the QuakeML format. Why don't you put these types in your sc3ml_0.10.xsd?

Because the ```sc3ml_0.10.xsd``` is generated from a more generic description while the ```quakeml_types.xsd``` is not.

> By the way, your modification works well! I can validate SC3ML files with it.

Great

gempa-jabe, 2017-12-07:
Can you please try this file? It is just one file I tried to generate. Does it work?

[xsd.zip](https://github.com/SeisComP3/seiscomp3/files/1539554/xsd.zip)

Brtle, 2017-12-07:
The defined type is  `FloatArray` but you use `RealArray` everywhere except in the `MomentTensorComponentContribution` node. You should use `FloatArray` or `RealArray` but not both.

```
>>> etree.XMLSchema(etree.parse('/tmp/sc3ml_0.10.xsd'))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "src/lxml/xmlschema.pxi", line 87, in lxml.etree.XMLSchema.__init__ (src/lxml/lxml.etree.c:196366)
lxml.etree.XMLSchemaParseError: element decl. '{http://geofon.gfz-potsdam.de/ns/seiscomp3-schema/0.10}probability', attribute 'type': The QName value '{http://geofon.gfz-potsdam.de/ns/seiscomp3-schema/0.10}RealArray' does not resolve to a(n) type definition., line 179
```

Note that I created a small SC3ML 0.10 from a SC3ML 0.9 file to test the XSD validation. It may contains some error with the TimeArray field. The `pdf` node looks like:

```xml
<pdf>
    <variable>2002-05-30T09:00:00 2003-05-30T09:00:00 2004-05-30T09:00:00</variable>
    <probability>0.5 1.3 2.8</probability>
</pdf>
```

gempa-jabe, 2017-12-08:
I have again reworked the XSD, find it attached. I also added your pdf to an origin time and it validates now.

[xsd.zip](https://github.com/SeisComP3/seiscomp3/files/1541851/xsd.zip)

Brtle, 2017-12-08:
It works for me too. Thanks for taking time for that!

closed 2017-12-08 by gempa-jabe