SeisComP3/seiscomp3 #157: Problems with single quotes at the beggining of earthquake name

jordi-domingo, 2017-12-07:
Hi all,

I'm experiencing what seems a bug when using single quotes in the earthquake name (or any EventDescription), specifically when the single quote is at the beginning.

i.e: if I put the string 'test as earthquake name in scolv, it is saved in the database as 'test'test

```
+----------+-------------+---------------------+------------+-----------------+
| _oid     | _parent_oid | _last_modified      | text       | type            |
+----------+-------------+---------------------+------------+-----------------+
| 38625521 |    38460201 | 2017-12-07 15:39:47 | 'test'test | earthquake name |
+----------+-------------+---------------------+------------+-----------------+

```

Seems to be a problem with escaping these single quotes when storing to the database (SQL).

It even gets worse if you try it more times without restarting SC3, because it stores the string repeated >2 times (and also with other strings previously tried!).

gempa-jabe, 2017-12-08:
I confirm this behaviour. It is not a matter of wrong escaping. An internal reset of a variable was just missing.

gempa-jabe, 2017-12-08:
It is fixed with 4f465942c4b532900cd86c4db858b2b14eb9a360.

jordi-domingo, 2017-12-08:
Great, it works!
Thanks for the quick fix.

closed 2017-12-08 by jordi-domingo